keirf/flashfloppy #1: Instructions

starfishmod, 2015-06-04:
Hello I bought one of these devices for some music equipment. I would love to replace the firmware as it's terrible. I can across the project and I would like to know more about it and the instructions for compiling and loading on to the device.

Thanks

omikron88, 2015-06-05:
Dne 5.6.2015 v 0:19 Andrew napsal(a):

>
> Hello I bought one of these devices for some music equipment. I would 
> love to replace the firmware as it's terrible. I can across the 
> project and I would like to know more about it and the instructions 
> for compiling and loading on to the device.
>
> Thanks
This alternative firmware is in the development phase currently, and 
cannot be used yet.

The original firmware is read-protected. This means it cannot be save 
for future use.
Flashing the alternative firmware is described at this page:

https://cortexamigafloppydrive.wordpress.com/

where the alternative software for Amiga computers can be downloaded.
Because Amiga uses it's own custom disk format, it cannot be used for 
PC's and other devices.

Omikron

keirf, 2015-06-05:
Hi, As Omikron says my Gotek firmware is quite far from completion. I am now targeting alternative hardware and may circle back to Gotek hardware later (the tricky bit being the USB stack, which my new target doesn't require).
Anyhow, that doesn't help you right now. Nor does the abovementioned Amiga-specific firmware. However, the author of the HxC2001 project (go look it up) has just released firmware for Gotek which would suit your needs. The downsides are: (1) you need to convert your floppy images into a new format, though there is a handy program provided for doing this; and (2) the firmware costs you 10 Euro. It is however your only replacement firmware option right now.
 Regards, Keir

closed 2015-06-05 by keirf

omikron88, 2015-06-05:
Tell me more about your hardware.

I wanted to build my own firmware for Gotek, because I want to use it as 
the floppy replacement for the historical computers. And many of them 
use FM, not MFM. But I stucked on USB too, I am still unable to 
read/write the sector from the pendrive.

My idea was to use traditional image files, dd if=/dev/fd0 of=000.img bs=512
named 000.img, 001.img... stored in Floppy directory.

If no additional informations, the image will be handled just like 720KB 
DS DD drive,
80 tracks, 2 heads, 9 sectors, MFM

If there is a text file with the same number and .cfg, it will read the 
geometry from it.
For example, 8'' IBM standard is

tracks=77
sectors=26
len=128 #sector length
heads=1 #single side
rpm=360 # for index freq calculation
speed=500 #bit clock
mfm=0 # 1=MFM, 0=FM

Omikron
Dne 5.6.2015 v 12:40 keirf napsal(a):

keirf, 2015-06-05:
Email me: my email address is in every source file.