keirf/flashfloppy #2: Hardware details needed

szg0000, 2016-12-29:
Hi!

Can this firmware replace the original Gotek read-protected firmware?
Can this firmware handle bigger pendrives/images than 1.44 MB?

keirf, 2016-12-29:
This firmware does not currently target Gotek hardware as it does not 
have a working USB stack. It functions on my own similar hardware with a 
touch screen and SD card reader.

keirf, 2016-12-29:
The schematic is in the hardware branch of this repository. See pcb/MM/150, pcb/LC/150, pcb/TB/161 for schematics and layouts in Kicad format for boards based on various STM32F103 dev boards.

closed 2016-12-29 by keirf

lovrenco, 2016-12-29:
Long time since then @keirf ... :)
Why not develop something for Gotek since you have the knowledge for it? Gotek is still produced and available on very affordable price.

keirf, 2016-12-29:
This firmware will be fairly easy to retarget to Gotek in due course as that is a more capable chip than STM32F103x. It will just need a USB stack to be added, probably ST's free one.

lovrenco, 2016-12-30:
Any chance to see that firmware soon?

keirf, 2016-12-30:
I need to knuckle down and finish the STM32F103 firmware, then I will look to retarget at Gotek. Such things are perpetually 'a few months away' as other repair work and so on keeps pushing development back.