keirf/flashfloppy #5: PC mode not working properly

BwanaFr, 2017-08-24:
Dear Keirf,

I've tested the v0.60a with my old Pentium II computer.
The mounted image is a HFE DOS 6.22 bootdisk (ok on HxC firmware).
Try to use the floppy in DOS. The DIR command failed by a General failure reading drive A.
I'm able to read the Volume serial number when I select Fail option (in the DIR command).

I successfully used this image one time (and one time only) but it was very slow...

Jumper configuration on Gotek is : 
- JC
- M0 (also tried with S1)

Thanks a lot!
Your firmware is the best :)

keirf, 2017-08-24:
Apologies, HD (1.44MB) disks are not yet supported, as I don't yet support that read rate off the USB stick. Of course this makes my feature list unclear as just about all PC disks are 1.44MB :) I will mention this limitation in the release notes. This will shortly be fixed in two ways:
1. I will support IMG files -- these require much less I/O bandwidth
2. I will improve read rate from the USB for HD HFE images (simply need to read sectors in larger batches).

keirf, 2017-08-24:
Actually I will leave this issue open to track the feature.

keirf, 2017-09-19:
New firmware v0.7.0a directly supports HD IMG files. Please try read and write with IMG files instead of HFE files. Open further tickets for any problems.

I will close this ticket as the feature (HD support via IMG files) is complete.

closed 2017-09-19 by keirf