keirf/flashfloppy #6: Is there Extended ADF support?

db260179, 2017-09-13:
Great work on this firmware for Gotek. After playing with some games etc, noticed that cannon fodder save disk needs an extended adf to save - 2mb rather than 880k adf.

I've used the tool xdftool to create a blank adf, workbench can format and is visible, but i believe the firmware is not understanding this extended adf?

Thanks
David

keirf, 2017-09-13:
You need to convert to HFE, which is supported. Actually it doesn't really matter if the HFE is initially blank I suspect. Why not take a copy of AUTOBOOT.HFE and rename it for your save-game image?

I will leave this issue open however, to track EADF support (which I will get round to eventually).

db260179, 2017-09-15:
Hi, ok thanks for the info. Yeh works now.

Jammmmm, 2018-03-12:
I'd like to +1 for extended ADF support :)

I'd love to see IPF support too, but I don't think that's going to happen anytime soon.

keirf, 2019-06-18:
I'm not going to do this. The few original games distributed as EADF (BC Kid, .... any others??) can be converted to HFEv2 via the usual method. Custom save games disks, just use a blank HFE image and format in game (eg. Cannon Fodder). You can make an HFE using Python script shipped with FlashFloppy:
```
python scripts/mk_hfe.py save_blank.hfe
```

closed 2019-06-18 by keirf