keirf/flashfloppy #7: Displaying more text with narrow font

zzarko, 2017-09-13:
Not an issue, more a small question and/or proposition... When 7x16 font already exists, is it possible to use it on full display width, to display more chars in adf/hfe name? Maybe even to display file extension (or maybe just an indicator, H-hfe, A-adf) in the second row of text, giving more space for the filename?

keirf, 2017-09-13:
Yes the 7x16 font will be integrated into the main firmware build and # char columns will be configurable via an INI style config file. Very soon :)

Leaving this issue open to track the feature.

keirf, 2017-10-11:
7x16 font is now integrated into the main build. Remaining work item is a config option to allow 18 chars per row when using the 7x16 font.

zzarko, 2017-10-11:
GREAT! Thank you!

bstrobel, 2017-10-17:
And while we are at it, displaying more lines/information on the very common OLED displays with 128x64 (as opposed to the 128x32 ones recommended here)?

Note: My 128x64 with SDD1306 work already with no changes, at least for me. The display is usable. It seems only that every other scan line is skipped and the characters appear somewhat stretched. Aside from that it is working fine though.

keirf, 2017-10-17:
Yes the question is: what to put on those extra lines. :)

tenuke, 2017-10-17:
demo effects :>

zzarko, 2017-10-17:
Maybe:
- displayng file name in two rows, so no more scrolling
- displaying the name of parent directory or full path (maybe combined with file name)
- current mode (native/autoboot/index)

bstrobel, 2017-10-18:
How about showing it like this?

```
  Indy3_1.adf
> Indy3_2.adf
  Indy3_3.adf
02/17 T:35 S:0
```
Indy3_2.adf is the select file. In the first line the previous file in the list is displayed, the current file in the second line (marked by "> "), and in the third line we have the next file. Status is in the fourth line.

bstrobel, 2017-10-18:
And instead of wasting space with the selection marker "> " you could also display the selected file in line2 inverse (black on white instead of white on black).

closed 2018-02-22 by keirf

zzarko, 2018-02-22:
Wow, Keir, thanks, I'll try this as soon as I can!

commit 73df00a5 (2018-02-22): oled: New display-type suffix "-narrow" to specify restricted view.
Increase 6x13 font chars-per-row to 21 or 18, depending on above.
Fixes #7