keirf/flashfloppy #8: ST images not being recognised in 0.6.2

v-retro, 2017-09-15:
This issue occurs on the Atari ST with 0.6.2, after assigning an ST disk image.
After 0.6.2 resolved an issue with writing selections from the HxC image selector, while the selections are now saved, rebooting from the allocated slot results in the drive being unreadable under TOS and therefore won't boot from the disk.

keirf, 2017-09-15:
ST images are not yet supported. You must convert them to HFE.

ST image support will be added soon, I will leave this issue open to track the feature.

v-retro, 2017-09-16:
Thanks.  HFE format works brilliantly.

keirf, 2017-09-19:
This is now fixed in v0.7.0a. Please test and let me know any issues.

I will close this ticket now as the feature is complete.

closed 2017-09-19 by keirf