keirf/flashfloppy #9: amstrad cpc dsk support

Breiztiger, 2017-09-16:
hi

very good job for gotek drive !!!

do you think supporting native dsk format for Amstrad cpc ?

thanks in advance

keirf, 2017-09-19:
Not sure, it's not a pure sector image like ADF/IMG/ST. It's not too far off though so probably supportable with not too much trouble.

I will leave this issue open for now to track progress.

Breiztiger, 2017-09-19:
if only no protected dsk should work, it's very usefull for demo or cracked games !!!

keirf, 2017-09-19:
Yes, I see no reason that shouldn't work. By the way every CPC DSK I've downloaded has been "Extended DSK", I wonder if there are any examples of "normal" ones?

Breiztiger, 2017-09-19:
as no protected dsk i think for edsk with only 40 track with 9 x 512k sectors, no gaps or fdc errors in st1 or st2

keirf, 2017-11-15:
This is done as of v0.9.0a. Please raise new issues for any problems with the DSK support.

closed 2017-11-15 by keirf