keirf/flashfloppy #10: Amiga ADF-support for various sizes ... HD-floppy not detected

fook42, 2017-09-21:
the issue derives from the new (>0.7.1a) feature of various adf-sizes.
amiga computers can deal with hd-floppys (capacity of 1.76mb), but an adf with this size is not handled as hd-floppy, because of a specific detection-mechanism on the floppy-interface.
-> to support hd-floppys on amiga, we need to implement this:
when the floppy is selected (select-line active), the ready-line has to "toggle" 10101010 .. instead of being "active" (=dd mode)

keirf, 2017-09-22:
There are a couple of things to do:
1. ADF handler must detect an HD image and switch to 22 sectors per track, but remain at DD data rate
2. This ID method must be implemented, if possible. First barrier is that the MTR line is not connected to the STM chip at the moment, needs to be jumpered to a spare input pin. Second barrier is that this pin may carry MTRX and need latching to generate MTR...

Does the ID need be implemented only for external drives? FF does not generate an ID signal for external DD drives either. 

If we do want to try generating the ID in software it might make sense to assume MTR is always asserted, and thus always generate the ID stream. Disadvantage for HD is that we wouldn't know if AmigaOS reads ID=101010... or 010101... I wonder if AmigaOS cares?

fook42, 2017-09-22:
> If we do want to try generating the ID in software it might make sense to assume MTR is always asserted, and thus always generate the ID stream.

wont work .. the ID-stream is only "read" when the MTR-Line is inactive.. when MTR is active, the ready-line has to follow "select" ...

> Disadvantage for HD is that we wouldn't know if AmigaOS reads ID=101010... or 010101... I wonder if AmigaOS cares?

i read, that it makes a difference whether the ID starts with 0 or 1...  010101  means some sort of identification for 5,25"-Drives (40Tracks) .. whereas 101010.. means HighDensity (1,76MB) ..
if ready is equal to select, then its 80 Tracks, 880KB.

just another information i found; the select-signal is set/unset serveral times before accessing a disk, an HD-drive will "respond" on the "ready-line" once ever 2nd-time (!) .. an DD-drive answers every time with ready = active!
(this is how this pattern is created on the Hardware-level)

keirf, 2017-09-22:
I don't believe "following select" when MTR is active is that important, as AmigaDOS traditionally ignores RDY (apart from drive identification!).

However I think you're right that we have to sync the ID stream to MTR sufficiently to ensure that the ID is read as 101010... 

Is this all needed for DF0, or just DF1+? Would it be okay to assume we have a valid MTR signal or only MTRx from which we need to create MTR as a function of MTRx, SELECT, and RESET?

keirf, 2018-05-01:
Now working for me. Try this firmware with `interface = amiga` in FF.CFG. This new interface type is a bit of a hack, I recommend it only for users who want to use HD images, and/or are having problems getting their Gotek recognised as an external drive.

[ff_10_3.zip](https://github.com/keirf/FlashFloppy/files/1963317/ff_10_3.zip)

closed 2018-05-01 by keirf