keirf/flashfloppy #11: Update in flashing instructions

TabajaraLabs, 2017-09-25:
Dear friend
Please, add these info to the flashing doc
http://tabajara-labs.blogspot.com.br/2015/03/amiga-gotek-floppy-emulator.html
This is a common problem found in flashing. The data/clock port are open collector and needs these two resistors to operate properly. Most USB to TTL adapters doesn't have it. 
THANKS for the software and it being open. Greetings from Brazil!

keirf, 2017-09-25:
Thanks. Not an issue I've seen but there it is in the official documentation so I will add a mention of it. :)

keirf, 2017-09-27:
Actually, before I do, have you any concrete examples of where this has been a problem? I've never seen the problem myself, nor had it reported to me by any users, nor does any other Gotek programming guide mention the pull-up resistors. Furthermore AN2606 explicitly mentions that Serial TX is configured as Output Push-Pull on STM32F105xx, and we can expect that typical serial adapters have *their* TX as Push-Pull too.

Hence I'm skeptical -- there are quite a few ways in which programming can and often does fail (bad host drivers, failure to negotiate the correct comms protocol, failure to negotiate the correct baud rate, ...) and explicit mention of this previously unknown issue risks confusing users.

TabajaraLabs, 2017-09-27:
Hi Keir

    Well, some 50+ flashed drives with many different usb-to-ttl adapters, 
count? :)

    I work with hardware design in Brazil, and I'm very used to these 
usb-to-ttl chinese adapters, not always they completely follow the 
specifications. Believe it or not, I had this problem in most drives I 
programmed to friends, for use in Amiga computers and old music keyboards. 
Feel free to use or not this info. I had problems and these pair of 
resistors were the solution :)

    Greetings from Brazil!
    Alexandre

---

<author@noreply.github.com>
Sent: Wednesday, September 27, 2017 11:45 AM

https://github.com/keirf/FlashFloppy/issues/11#issuecomment-332545248

---
Este e-mail foi verificado quanto a vírus pelo AVG.
http://www.avg.com

keirf, 2017-09-27:
Okay, any chance of a photo of how you attached the pullups? I guess you solder them on the adapter itself. Also this would give a nice picture of an offending adapter :) Most people are just gonna run scared of being given a schematic like in the AN2606 document they need to be shown physically solder point A and point B ;)

keirf, 2017-09-27:
No worries, I will include this subject as part of a larger troubleshooting section on serial programming.

Povvercrazy, 2017-10-09:
Hi Keirf, I have had this exact problem to say the least it was extremely frustrating, I am using a CP2102 chipped usb-ttl module and it just wasn't working, found this thread, added the two resistors and worked perfectly, might be a very small amount of people who suffered this issue, but I think its worth mentioning for the few who may encounter it. Thanks for your work btw !!

keirf, 2017-10-09:
Okay this is mentioned in a new Troubleshooting section, with link to AN2606 app note.

closed 2017-10-09 by keirf