keirf/flashfloppy #12: Larger Rotary circuit diagram

smartroad, 2017-09-25:
Hi!

Loving the firmware. Is there a larger version of the image for the rotary encoder? I can't read the size of the resistors 😞 I am assuming they are 1K but just in case I am wrong!

https://github.com/keirf/FlashFloppy/blob/master/doc/assets/rotenc.jpg

Cheers!

keirf, 2017-09-25:
You do not need pullups they are just part of the Ky040 module schematic. In fact the pullups can make the encoder inputs unreliable so you are better without them.

I will update the documentation to mention the resistors and that they should be ignored.

fook42, 2017-09-25:
i will send a new picture... with the additional connection for the Push-Button (Button #3) .. and the +3.3V - Line.. okay?

keirf, 2017-09-25:
Yes sounds good thanks. I am going to move the docs to the Github wiki by the way, but that won't change anything much, just moves docs and assets out of the source repository into *.wiki.git repo instead.

fook42, 2017-09-26:
here is my proposal: 
[image]

smartroad, 2017-09-27:
That's great much more readable thanks!

keirf, 2017-09-27:
There are a couple of regressions compared with the original image:
1. Pin B is no longer labelled
2. The colour codes are no longer repeated at the encoder pins themselves. Clearer if you add a short span of colour by each pin: A=red,B=blue,C=brown,switch1=green,switch2=brown

Please can you make these fixes and I will then commit the picture to the new wiki pages.

Cheers!

fook42, 2017-09-27:
you're right, Keir... i took your advice and made the changes... the labeling of the encoder is not of any use i guess, so i removed it completely.
[image]

keirf, 2017-09-27:
Sorry, I think the A/C/B labelling at least is useful as it matches up with those pin names in the instructions. Could you add them back?

Apart from that -- looks super good, much better than my own pics ;)

fook42, 2017-09-27:
next update...
[image]

keirf, 2017-09-27:
Can't argue with that! Thanks.

closed 2017-09-27 by keirf