keirf/flashfloppy #14: Using FlashFloppy on a Robotron Computer

danau, 2017-09-30:
Hello Keir,
with joy I follow your developments from FlashFloppy and have also tried every release. You're doing a great job.
I use a robotron computer which has a CP/M 2.2-based operating system. Since version ~ 0.6.x works with the reading and writing on the virtual disk. Unfortunately, I have not managed to boot from this image.  I have used the configuration-less mode. Do you have an idea what I do wrong?
Thank you and best regards
Spaceman

keirf, 2017-09-30:
Kingstener has reported same boot issue on cp/m. The issue is almost certainly side select handling. FlashFloppy flushes it's buffers on a side change and takes milliseconds to recover. Most hosts don't care but cp/m loader seems to twiddle the side select a lot and expects better responsiveness.

I may fix this for HFE prior to v1.0. Other image types later as the required interleaved reads need a bit more care to not hurt performance (HFE image format is already interleaved).

keirf, 2017-10-06:
Kingstener points out that the problem is that the CP/M loader is written for compatibility with very old single-sided 8-inch drives which use the Side-Select pin for Fault-Reset instead. Hence this line gets asserted for ~16us presumably when about to read a sector. This of course causes FlashFloppy to go away for sufficient time that the sector is unreadable every time.

Current suggested quick fix is to filter short changes on side-select as 'glitches'.

The longer-term fix is more work. Definitely post v1.0.

danau, 2017-10-07:
Hello Keir,
thank you for your message. Also thanks to Kingstener for finding the problem.
Can you incorporate the fix of Kingstener -to the complete elimination- into your next firmware versions? I think that helps a lot of people.
Many Thanks!
Spaceman

keirf, 2017-10-07:
That is the plan, I await initial patch from Kingstener then I will revise it with him for inclusion into the following release.

keirf, 2017-10-12:
Fix is applied to master and will be rolled into the next release.

closed 2017-10-12 by keirf