keirf/flashfloppy #16: Acorn device support

fordp2002, 2017-10-10:
I got my first drive today and would like to use it on my BBC Micro's so would like to add support for SSD and DSD files. I think they should be very similar to the Atari formats. An SSD is a single sided 80 Track 10 Sector single density disk with 200K Byte capacity.  The DSD is the Double sided 400K Byte version. There are also ADFS formats that can be added at some point too.

I will have a go at doing this myself in the first instance but would like at least to be able to add to the Wiki.

Thanks.

keirf, 2017-10-10:
If single density then you need to generate FM rather than MFM. This has a few differences:
- FM clock bits are always set
- Gap bytes, gap sizes, some mark bytes, etc are different

Feel free to try it out, my email address is in the Git commit logs if you want to discuss.

fordp2002, 2017-10-26:
I just thought I would post an update. I am slowly working on DFS support and am posting my work in my fork. I am understanding how floppy drives work more all the time and have add the i8271 document to my repository as it is full of great information. I only spend a few hours a week so it will take a while yet!

drdpj, 2017-11-06:
Ford, old chap :D - it's danielj from stardot. My gotek is due to turn up any day now. I did quite a bit with the kryoflux and decoding bbc micro FM streams in java, so I'll dive in on your fork if that would be helpful? (if I can work out this githubby thing) :)

keirf, 2018-02-01:
Chaps, If you have a BBC with DFS 1.20 (not the patched DFS 1.21) can you try HFE images on the latest release v0.9.7a. See #47 for context, and a (possibly partial) fix.

I may also pick up the SSD/DSD work...

fordp2002, 2018-02-01:
Sorry I have not got round to doing this. I must admit when I said I would do this I assumed there would be an existing FM implementation (Like PC 360K Floppies) to hack and it would just involve the DFS specifics. I also expected the hardware would provide SW debugging but it seems the two pins needed for that do no even go out to test pads. I still hope to help in some way. DFS is unusual as the two sides of the disc are often treated like separate drives.

I will try and have a go this weekend to at least flash my drive with the latest version!

keirf, 2018-02-01:
It's not a problem, I've been meaning to get round to doing FM formats for a while. I will post a firmware here for test in due course.

drdpj, 2018-03-28:
Is there anything I can do to help this along - my workbench is clear for a few days and I should be able to put a couple of hours in here and there?

keirf, 2018-03-28:
Thanks. Quick question: are DSD files always 200k (40track) / 400k (80track)? Are SSD files always 100k (40track) / 200k (80track)?
I wasn't sure if zero sectors at the end of the image are truncated?
Are there some example images I can download?

keirf, 2018-03-28:
Also, should any interleave or skew be applied to sector ordering?

drdpj, 2018-03-28:
Sometimes they're truncated if later sectors are blank... I don't think
this is very useful, however that's just history... The suffix is always a
reliable indicator of the format though.

drdpj, 2018-03-28:
Sectors are ordered normally in the images, but tracks are interleaved between sides in DSD images. I can't quite remember whether there's any skew on real formatted discs, but there are a few HFEs that were uploaded in the other thread that should give it away.  Lots and lots of SSDs here: http://www.stairwaytohell.com/essentials/index.html?page=homepage

keirf, 2018-03-28:
Ok, I'm working on read-only support for now. Will have a firmware to try, later today.

keirf, 2018-03-28:
Okay here is prelim read-only support for SSD and DSD images. Please test. :) Also interested in performance, as this currently has no special sector interleave or skew.

[ff_16_1.zip](https://github.com/keirf/FlashFloppy/files/1856602/ff_16_1.zip)

drdpj, 2018-03-28:
OK - that works perfectly on the Master with 1770 - it doesn't behave on the Beeb with 8271 controller - Disc Fault 18 (Sector not found) - the latter can be caused by timing being off...

drdpj, 2018-03-28:
Speed-wise it's a bit slower than a real drive - verify 80 tracks takes 34 seconds against 23.

keirf, 2018-03-28:
Can someone attach an HFE that works full speed and works with 8271 controller?

drdpj, 2018-03-28:
[Watford Diags_dsd.zip](https://github.com/keirf/FlashFloppy/files/1857101/Watford.Diags_dsd.zip)
This works at the same speed but is fine with the 8271 (I tried verifying on both the 8271 and 1770, and it was 34 seconds).  I can kryoflux and upload a master formatted disc converted to HFE in a mo.

keirf, 2018-03-28:
The format produced by FlashFloppy was already very close to the Watford Diags HFE. I have tweaked it to make it even closer, but I'm not sure whether the problem must lie elsewhere... Anyway I attach it for you to try out.

Is there a head-settle time that the BBC 8271 expects, after a seek operation?

[ff_16_2.zip](https://github.com/keirf/FlashFloppy/files/1857382/ff_16_2.zip)

drdpj, 2018-03-28:
OK - that fixed it! :)

drdpj, 2018-03-28:
This might be useful: http://msknight.com/bbc/manuals/advanced-disk-user-guide.pdf

drdpj, 2018-03-28:
[8271DataSheetpdf.pdf](https://github.com/keirf/FlashFloppy/files/1857472/8271DataSheetpdf.pdf)
Also for good measure...

keirf, 2018-03-28:
Ok, that firmware added extra space after the index mark, which is not in the Watford HFE nor in the 8271 datasheet. Let's see if this one suffices (keeps reduced gap between sectors, but gets rid of the non-standard extra gap at track start):

[ff_16_3.zip](https://github.com/keirf/FlashFloppy/files/1857493/ff_16_3.zip)

keirf, 2018-03-28:
Ah ok so ff_16_2 didn't work so well after all. Disc Fault 10 is insufficient index pulses. You are running `index-suppression = no`?

drdpj, 2018-03-28:
Also fine :)

keirf, 2018-03-28:
Out of interest does 16_1 now work too, or is that still duff?

drdpj, 2018-03-28:
Also fine. For shame *red faced*

keirf, 2018-03-28:
Haha ok. Leaving as in ff_16_1 then, and now working on write support. The only tricky bit is syncing the bit stream on the address marks.

drdpj, 2018-03-28:
I'll see if I can get some of the other beeb-lot to wake up and follow this issue. The more testing the better really...

keirf, 2018-03-28:
Here's a stab at write support (read support is as in ff_16_1):

[ff_16_4.zip](https://github.com/keirf/FlashFloppy/files/1857799/ff_16_4.zip)

drdpj, 2018-03-28:
Just been recalled to reality, so will have to give that a shot tomorrow...
What happens with truncated images? Does it extend them? Or do you need
full size if you want to write with impunity?

keirf, 2018-03-28:
Truncated images are not extended: writes outside the image are silently dropped. So beware ;)

kgl2001, 2018-03-29:
Duly woken up, and here to help! I'll pick up this latest test image ff_16_4 this evening and report back. I should be able to test on both 1770 & 8271 model Bs.

keirf, 2018-03-29:
If you see any problems with reads (specifically the error 18) then please also test read-only firmwares 16_2 and 16_3.

kgl2001, 2018-03-29:
With ff_16_4, `index-suppression = no`, 8271 controller, DFS1.20 and testing with Watford Utilities:

- Formatting working fine, but get occasional 'fast' stepping (0..1..2..3.4..5..6..7..8.9..10..11..12..13..) during track changes. Doesn't cause any errors. I think I was also seeing this with the .hfe image
- Verify working fine
- Write tracks 0..79 working fine
- Read tracks 0..79 - fails at track 79 (read error)

So far I've run this test twice with the same error.

Edit: Make that 3 from 3 read errors at track 79.

kgl2001, 2018-03-29:
Good shout! It was a 388k file. It should be 400k! Let me retry with a fully padded .dsd file. As a refinement, perhaps worth ignoring .dsd files that aren't exactly 400k or .ssd files that aren't exactly 200k (for 80trk images - not sure I've ever come across 40trk images). I know this is a question you asked the other day.

Edit: Got past the read error at track 79 now that I'm using a fully padded .dsd file.

kgl2001, 2018-03-29:
That's me completed one full set of Watford Utilities tests and one full set of Disk Utilities tests, and both have passed without any errors 😄.

I've got a suspicion that read and write are both slower than the real floppy, but I will need to test fully to verify this.

Running out of time tonight for further tests, but I'll pick up with further tests on the 8271 and kick off some tests on the 1770 tomorrow.

keirf, 2018-03-30:
Regarding speed: one thing worth checking if you can is interleave/skew on those real disks. However that would require imaging to HFE and inspection in the HxC track analyser. Skew means that sector 0 is not always the first sector in the track -- instead it gets offset from track to track. Interleave means that logically adjacent sectors are not physically adjacent within a track -- eg. interleave=2 means sectors are 2 sectors apart rather than 1 (1 = adjacent).

Regarding silent drop of writes to truncated images: this can be improved for direct-navigation mode (ie. not using HxC file selector, which doesn't exist on Acorn platforms anyway). It is a separate work item however so please raise a separate ticket for this issue.

drdpj, 2018-03-30:
Here's a BBC formatted disk converted to HFE - no skew that I see.
[blankdfs.zip](https://github.com/keirf/FlashFloppy/files/1863020/blankdfs.zip)

I'm getting watford and the other disk test bombing out on track 79 with a read error (I think - watford doesn't stay on the screen long enough to see what it's doing) - looking at the blank dsd file that it's messing with, I can't see anything odd - it's all 0s...

keirf, 2018-03-30:
@drdpj Is the DSD file full size (409600 bytes) or truncated?

drdpj, 2018-03-30:
Full size.
[blankdisk.zip](https://github.com/keirf/FlashFloppy/files/1863065/blankdisk.zip)

keirf, 2018-03-30:
Might need logging from a debug build then. Can you do that, off `dsd ` branch?

drdpj, 2018-03-30:
On it...

drdpj, 2018-03-30:
Irritatingly it seems now to be working fine. I'll push it through a few more iterations over the next few hours...

drdpj, 2018-03-30:
OK - 0 problems with the debug build I just did. Possible that timings are slightly tweaked by the serial output?

keirf, 2018-03-30:
Do your own non debug build and try that

drdpj, 2018-03-30:
No issues - my non-debug version attached.
[flashfloppy_9.16-4-DJBuild.zip](https://github.com/keirf/FlashFloppy/files/1863482/flashfloppy_9.16-4-DJBuild.zip)

keirf, 2018-03-30:
Well that's weird. I can't actually remember if 16_4 directly corresponds to tip of `dsd`. I will pop another non-debug build here now (ff_16_5). Other thing I thought of -- did you use a fresh DSD every time you start the test?

[ff_16_5.zip](https://github.com/keirf/FlashFloppy/files/1863515/ff_16_5.zip)

drdpj, 2018-03-30:
No, same one. It formats it as the first part of the test. That bit always
works...

keirf, 2018-03-30:
Well see how ff_16_5 goes.

kgl2001, 2018-03-30:
I'll try it when I get home shortly, but what's new in 16_5?

keirf, 2018-03-30:
@kgl2001 was addressed to @drdpj mainly, who had problems with ff_16_4 but not with his own builds. So ff_16_5 is a build that is definitely up to date with the `dsd` development branch.

drdpj, 2018-03-30:
All good on that one!

keirf, 2018-03-30:
Turns out ff_16_4 was already up to date with `dsd`. ff_16_4 and ff_16_5 are absolutely *identical*. So, I don't know what is/was up with your setup @drdpj  :)

kgl2001, 2018-03-30:
ok. so I won't bother with 16_5, and I'll keep testing with 16_4.

keirf, 2018-03-30:
That is totally fine yes @kgl2001

drdpj, 2018-03-30:
No idea :-( It's got a dirty big old ROM board in it but nothing else. It was all fine on the Master. No matter, it must have been the machine being flakey rather than anything else. I've not been able to break it again.

kgl2001, 2018-03-30:
Done another test on 8271 using disk utilities, and it all passed again.

Just tried to test on another beeb with the 1770 controller, and both test utilities are throwing errors. I guess they're not compatible with the 1770? I need to do a bit more investigation.

keirf, 2018-03-30:
@kgl2001 Do regular read operations work on pre-existing DSD images with ff_16_4?
If even that does not work, can you try read operations on ff_16_2 and ff_16_3?

kgl2001, 2018-03-30:
It was the format operation that was failing when running the Disk Utilities / Watford Diagnostics tests. The format and verify commands that come with DFS2.26 work fine, so I guess the different controllers need different commands for the format to work.

I can bypass the format / verify tests in Disk Utilities, and have moved onto the write / read tests. These tests are still running on a .dsd disk, and initial indications are good (still using ff_16_4).

keirf, 2018-03-30:
Oh I see, perhaps the DiskUtilities / Watford format tests expect to format as double density? I'm no expert on these BBC systems and how easy it would be to test FM / single-density on ADFS systems.

[21 more replies]

closed 2018-04-05 by keirf