keirf/flashfloppy #17: Hardware Mod LCD. Led turn on the blue screen ..but no text comes out

Starvak, 2017-10-14:
Halo.
I have restored a veteran ST, and I wanted to try the gotek emulator.
Google has brought me to this fantastic project. I have flashed the gotek, and it works. I put a beeper on it, and it rings. I bought an lcd exactly the same as the webpage, .... but it does not work correctly. The screen turns on, but no file text appears. any solution?

keirf, 2017-10-14:
Should be a small variable resistor on the back adjustable with a small screwdriver. Try turning it.

Starvak, 2017-10-14:
I will try to vary the resistance. Thank you.

But I have to complete the query. The display is not exactly the same. It has three open configuration bridges.
I found this. It seems to set the ic2 address. Chip 8574AT
Range: 0x38; 0x39; 0x3A to 0x3F
Could it be that?

keirf, 2017-10-14:
No should work at any address.

Starvak, 2017-10-14:
Well, adjusting the resistance, I see something on the screen.
Only blocks, filling the display.
I have three dispaly. Physically the same, maybe something more modern.
I have prepared them with their corresponding resistance 4.7k, another with 10k (as the photo of the web), and another without resistance. All present a line full of blocks.
The game starts well from the usb

Starvak, 2017-10-14:
Ups.... I try whit other gotek board and the lcd with no resistance, and ... working!
Now , I can work with this, to look for fault in the others.

Starvak, 2017-10-14:
Conclusions; Everything works well. (more or less)
In effect, the adjustment of the variable resistance, provokes saturation and presence of blocks of cells, (instead of letters)
In my lcd display, the incorporation of resistors does not have any effect.
Values tested 4,7k.omh and 10ohm.

There is a small detail with the gotek led. I have a red led next to the green. Without display, they act with activity. But with the display switched on, the red LED lights permanently, and green has almost no activity.

keirf, 2017-10-15:
The red LED denotes display activity. With LCD and OLED displays the display is updated regularly even when there are no changes. Hence the red LED stays lit. The red LED is quite pointless, if you don't like it you can always remove it or at least cut one leg of it inside the Gotek.

closed 2017-10-15 by keirf