keirf/flashfloppy #18: Layout image for Gotek

smartroad, 2017-10-22:
Don't know if this would be of any help. I made a basic layout image of all the major parts and ports/pins of the Gotek to help me remember where bits are (I find pictures useful but sometimes confusing as parts merge with the background to my eye at times).

I could use it to update the wiki wiring diagrams if it would be helpful for a more "schematic" overview as well as photos?

[image]

keirf, 2017-10-22:
A new top-level page "Gotek Block Diagram" under Initial Setup? Do you want to do that, or else I can take your diagram and incorporate it into what will be a pretty short page?

smartroad, 2017-10-22:
I was thinking as additional to the exisiting instructions. For the rotary encoder for example:

[image]

Hope you don't take this as presumptuous! I just wanted to help and not knowing that much about coding this complex, this is something I can offer for the cause 😄

keirf, 2017-10-22:
Oh I see. That's great! You can feel free to attach pictures here for me to pull into the wiki, or make wiki changes for me to pull in directly. Very happy either way!

keirf, 2017-10-22:
Your diagrams can also replace some of the existing pictures, that's fine too, if they're strict improvements.

tenuke, 2017-10-22:
very nice work! propably quite helpful for ppl like 0% technically oriented
who are still interested in adding the mods :D

tenuke, 2017-10-22:
oops I forgot to add that there could or maybe should then also be some
info about how they can identify if their rotary encoder does have the
pull-up resistors or not.
as these pics are after all aimed for peeps who cant figure out the correct
connections just like that?

2017-10-22 19:12 GMT+03:00 mikko nurmesniemi <nukester@gmail.com>:

smartroad, 2017-10-22:
Hopefully I have done it right. I have updated the Wiki with the rotary encoder (one image for 'naked' and PCB carrier options) as well as the pizeo sounder. Not added to another repository before so no idea if I got it right!!

keirf, 2017-10-22:
I don't see the changes, unsurprising as I don't think anyone else has permission to edit the wiki.
Do you have a private fork somewhere?

smartroad, 2017-10-22:
I tried to fork it but it doesn't copy the wiki over. I've been able to clone the wiki to my desktop client but it obviously won't let me push the changes as obviously I do not have any permissions.

keirf, 2017-10-23:
I'm not sure what the best solution is for merging local edits of the wiki. It doesn't seem well supported by Github. Maybe just zip up your modified version of the repository and upload it somewhere I can grab it.

fordp2002, 2017-10-23:
You can allow the wiki to be edited by any registered github user this what a lot of projects do.

keirf, 2017-10-23:
Yeah, I'm reluctant though. I thought I could add collaborators but that applies to the whole project, code and all... so it does seem it is all or nothing for permitting wiki write access. And no interface for viewing and merging wiki updates from third parties. That's crap, Github.

fordp2002, 2017-10-23:
The wiki is a separate git repository so it is pretty simple to roll any changes back. I accidentally deleted a page over the weekend on another project and got it back from git (hopefully before anybody noticed). Why not open it up for now and close it if you get a problem?

keirf, 2017-10-23:
My other concern is that opening up access permits access only via the direct online page editor, and that this wouldn't allow smartroad to add his pictures to the repository (which as I understand it requires the wiki repo to be cloned locally, modified, and updates pushed back). I'm happy to be put right on these points however, I may have misunderstood something.

smartroad, 2017-10-23:
Its really annoying as I have a clone of the Wiki sitting on my harddrive now and GitHub Desktop ready to upload the couple of modifications I have done. Probably then the easiest way is for me to make a new blank repo and then use its wiki to make my changes, you can then clone that one and upload to your one :grin:

keirf, 2017-10-25:
Let me know how it goes. I could try temporarily opening access to the wiki to see if you can then push your changes directly?

smartroad, 2017-10-26:
Sorry for the lack of response, I have been called away for work and forgot to transfer the images from my home machine to my laptop so can't work on them until I get home. I'll get it all done and then upload to a repository, you can then copy and transfer. Probably the easiest way without having to open up your repo completely :)

keirf, 2017-11-06:
Did you find time for this? I'm happy to try opening the wiki or get the assets from you in whatever form. :)

smartroad, 2017-11-06:
https://github.com/smartroad/temp-FlashFloppy-wiki/blob/master/Hardware-Mods.md I have made a start here. Annoyingly work has been keeping me away from it. I hope to have the mods page finished this week tho

keirf, 2017-11-06:
Ah looking good :)

smartroad, 2017-12-18:
Finished the hardware mods page: https://github.com/smartroad/temp-FlashFloppy-wiki/blob/master/Hardware-Mods.md#rotary-encoder let me know if I have missed anything

keirf, 2018-01-08:
Thanks I have now merged these new diagrams into the main wiki.

closed 2018-01-08 by keirf