keirf/flashfloppy #23: OLED display size pcfg option

Kyp069, 2017-11-23:
I have seen a note in the sources about changes needed to use an OLED 128x64 display but as I cannot rebuild the sources I would kindly ask for a option in the config file to specify what kind of display are we going to use. Thank you very much.

MartinDejmal, 2017-11-23:
I tried 128x64 OLED display and contents displayed on the OLED remains the same, but is streteched to fill the whole screen. 
However why use bigger display? What else should it display other than image filename and sector/track?

fordp2002, 2017-11-23:
A feature that will probably be needed for the DFS on the BBC and will be useful for other platforms too is having multiple images at the same time for drives and in the case of DFS sides of a disc.

Kyp069, 2017-11-23:
My OLED display shows a first _scrolling like_ line with something that appears to be corrupted characters and the rest random pixels most of them ON.

There is not a real reason to use a bigger display, just it is the one that I have :)

keirf, 2017-11-25:
Larger display is "supported" in teh way MartinDejmal describes. I will try to do more with the extra space in future. I can leave this ticket open to track it.

Kyp069, 2017-11-27:
This is a snapshot of what my display shows: 

[image]

The top line es 'moving' like it was doing some kind of scroll but text is illegible.

Perhaps I am doing something wrong?

keirf, 2018-06-20:
@Kyp069 This is fixed in v0.9.22a, put `display-type = oled-128x64-sh1106` in FF.CFG.

Kyp069, 2018-06-21:
I can confirm it works 👍 
Thank you for the support :D

closed 2018-07-24 by keirf