keirf/flashfloppy #25: file order

kodak80, 2017-11-25:
I am using my Gotek with OLED display in an Atari ST and loving this firmware.

The file/folder order seems to be the order copied onto the USB disk?  Would it be possible to have a FF.CFG setting to control the order that files and folders are shown?  ie. Alphabetically or the order copied onto the USB stick?

keirf, 2017-11-25:
This is not straightforward but... maybe as a feature post 1.0. It could be done, and maybe without slowing down directory listing (too much).

I will at least leave the ticket open for now to track this.

kodak80, 2017-11-25:
Thank you.  It was more of a nice to have so yes, appropriate that it will be a feature post 1.0 release.

Quantum-cross, 2017-12-19:
I haven't tried it with the Gotek yet, but try http://fatsort.sourceforge.net/

I've had to use this for my PowerPak NES flash cart to be sorted properly.

Simone2018, 2017-12-19:
I exactly need something like this in DOS or Win...

kodak80, 2017-12-19:
As it works under Linux, I might try creating a virtual Linux machine and mount my USB stick within it.

tenuke, 2017-12-19:
googled it for you ;)

http://www.anerty.net/software/file/DriveSort/

*http://www.compulsivecode.com/project_copyinorder.aspx
<http://www.compulsivecode.com/project_copyinorder.aspx>*
<http://www.anerty.net/software/file/DriveSort/>

Simone2018, 2017-12-19:
Thank you tenuke! I was sure that exist!

keirf, 2019-04-23:
Fixed in v2.7a

closed 2019-04-23 by keirf