keirf/flashfloppy #26: issues booting OS/2 install image

davide125, 2017-11-29:
I'm trying to use FlashFloppy on a gotek to install OS/2 Warp 4.52 on an old PC. The disk image (converted to HFE) seems fine, as I can read its contents correctly from a running DOS system, but it fails to boot with a SYS02025 error, which according to http://bos.asmhackers.net/docs/ata/docs/ALLHIW/hiwos2.HTM means that the INT 13 AH=00 disk reset call the bootloader issues is failing. I'm out of my depth here, but could this be due to something missing in the emulation that FlashFloppy does? Happy to provide more data or a copy of the images I'm using if needed.

keirf, 2017-11-29:
Seems more like INT 13 AH=02 (disk read sectors) is failing. I guess the hardware setup is good enough that the boot sector gets read and executed okay.

Was the original image XDF? You could email me that and the HFE and I could take a quick look. Perhaps it didn't convert quite right.

davide125, 2017-11-30:
I don't think disk 0 is XDF, though I could be wrong; the images are meant to be restored with loaddskf, which incidentally also doesn't work (it gives a generic error when trying to write to the gotek, both when run from MS-DOS 6.22 and from FreeDOS), but they look like regular raw image files. I put the whole thing at https://www.dropbox.com/s/36tzi6ve90185r1/os2boot.zip?dl=0 (original images, HFE, and a copy of loaddskf in case you can find it useful) if you wanna poke at it. Thanks!

davide125, 2017-11-30:
Also tested trying to boot the original image straight (without converting it to HFE, just by renaming the DSK to IMG), and it behaves exactly the same (errors out with SYS02025).

claunia, 2017-12-01:
@davide125 SYS02025 is a generic error with int13. It may be something of timing. You commented on facebook that you get a spurious a/r/f error with normal DOS operations, could you elaborate the exact message? (there are about a dozen of int13 errors that make DOS show a/r/f). It may be related to the problems I'm seeing.

davide125, 2017-12-01:
I saw fairly consistent a/r/f errors, for example, when trying to install WfW 3.11 from a HFE-format image; it would fire that when trying to read SETUP.EXE, but would otherwise work fine. Using the original image (in IMG) format avoided that error altogether. I also saw occasionally saw a/r/f when doing things that write to a floppy (e.g. stuff like COPY), but this seemed sporadic and non-deterministic.

Simone2018, 2017-12-01:
I also see a/r/f errors on writing (copy) to HFE (testing v1 & v3) but also IMG images created from scratch. Using "retry" don't work, but the "Ignore" option seems to bring to a correct writing operation. booting from a know valid floppy has success. Using MSDOS 6.22 - NEC V20 processor WD37C65 floppy chipset (720Kb DD density floppy only). 
One more thing... putting the Gotek in Direct Access Mode (track 255) require always 2 consecutive INT13 operation. The first has always Carry Set, so an error is occurred.
[image]

keirf, 2018-01-26:
Can you try again with this recent build of latest master?

https://github.com/keirf/FlashFloppy/files/1663928/ff_44_2.zip

davide125, 2018-01-27:
Thanks @keirf! I'll give this a shot when I'm back from travel (in a couple of weeks) and report back.

keirf, 2018-01-28:
No problem. I will shortly release v0.9.7a so by the time you are back you should just test that as it will include all the necessary improvements.

claunia, 2018-01-28:
pretty much seems to be the same problem I was getting, so I'll test it asap also.

keirf, 2018-02-07:
Anyone had a chance to do some testing yet? :)

davide125, 2018-02-07:
I'll get you something within the week, promise :)

keirf, 2018-02-08:
Another thing to mention -- if HFE still doesn't work, try the DSK images directly. They should now work, and more reliably than HFE at high density (HFE is better than it was but there's more work for me to do there).

davide125, 2018-02-11:
Using the latest release (v0.9.8a): the OS/2 HFE images fail in the same way, while the original ones work better: disk 0 reads and boots correctly, asks to insert disk 1, then after a bit it asks for disk 2, but as soon as that is switched it fails with "OS/2 is unable to operate hard disk or diskette drive".

davide125, 2018-02-11:
Trying to write out images with loaddskf against the gotek also still fails. I was able to install WfW 3.11, but I hit a handful of a/r/f errors throughout the install; for disk 8, I had to switch to the HFE image as it was consistently failing to read NET.EXE from the IMG, but in the end I made it through. In general, it feels the new version is a lot more reliable, but there's still a few issues.

keirf, 2018-02-11:
Thank you very much for this testing. Let's address these one by one:

1. HFE HD read is still unreliable. To stream predictably off the USB stick at sufficient rate I need a buffer cache. I will get this implemented post v1.0.

2. IMG HD read should be reliable. Interesting that it failed immediately on disk 2. Perhaps that is a different layout or sector numbering or something. Can you send me the IMG and HFE images for the OS/2 install set? You can email them to me if you like (my email is in my Github profile). Are the images (IMG set and/or HFE set) known/tested good?

3. Write is stuffed on latest firmware, since I implemented write pipelining. I have some testing to do and will get this fixed asap. So let's put this to one side for now. Were you writing to IMG or HFE, by the way? And do you know what mode loaddskf was running in (including format and/or verify for example)?

4. Failure on WfW disk 8 IMG file is a bit concerning. Are you sure the image file is good? As with OS/2 can you send me the IMG and HFE install sets, and confirm if either/both is known/tested good?

Thanks for your efforts! If we can iterate on this, drill into the issues, and try out some fixes, we can get these issues nailed pdq I'm sure.

claunia, 2018-02-11:
@keirf since OS/2 3, disk 2 onward are in XDF, a format that basically is 4 sectors per track. This should not be a problem for HFE, but for IMG, you need to add specific support, because the IMG appears like a 21 sectors / track with a completely different layout (XDF stores only 1 FAT, but the IMG has 2). I can send you a kryoflux of a real XDF from OS/2 3 plus the XDF file corresponding to it.

keirf, 2018-02-11:
@claunia yes please. Not sure if I can then autodetect between the layouts or will add a host= hint or a filename hint (eg new extension)

davide125, 2018-02-11:
OS/2 images + loaddskf: https://www.dropbox.com/s/36tzi6ve90185r1/os2boot.zip?dl=0
WfW 3.11: https://www.dropbox.com/s/0dnvorekbmgnfou/wfw311.zip?dl=0
I'm positive the OS/2 images are good (as they come from the CDROM); I don't have a good way to validate the WfW ones, but they do seem correct from a quick inspection. Unfortunately I don't have a real floppy drive around to actually test these on actual floppies.

keirf, 2018-02-11:
@davide125 Thanks. It's a bit weird that all three OS/2 sector images (suffix DSK) have the same size if disk 2 is in fact XDF. I will have to check that out.

keirf, 2018-02-11:
@claunia Any comment on davide's os/2 install set? Looks like it could be three regular DOS-format disks to bootstrap a CD install?

keirf, 2018-02-11:
@davide125 have you successfully used those OS/2 boot disks on this particular test system? eg. by XDFCOPY to real floppies and use a real floppy drive? It strikes me the error "OS/2 is unable to operate hard disk or diskette drive" doesn't necessarily mean the floppy access is the problem.

davide125, 2018-02-11:
No, though I did use them years ago on a different machine and that was fine. You're right, it's entirely possible there's something else going on here. I suspected the floppy part as when it shows the prompt to insert disk 2, I can hear the gotek buzzer keeps going; as soon as I switch the disk I immediately get that error, without even having to press ENTER on the keyboard.

On a different note, I tried a Warp 3 diskette install set, and from disk 2 onwards is fails with "err 31" on the oled -- as @claunia said, these are 1.9MB XDF images.

keirf, 2018-02-11:
@davide125 Ok that sounds interesting, maybe I will have to try to repro this myself. Gotek buzzer keeping going is interesting. Does it click, click, ... or is it a continuous grinding?

I would like to support XDF images. Did they just have a normal IMG/DSK suffix or do they go by an XDF suffix, conventionally?

davide125, 2018-02-11:
All the OS/2 images are suffixed DSK; I rename them to IMG before putting them on the USB stick, as I remembered having trouble when they were named DSK the first time I tried this back in November. I don't see an issue with requiring a XDF suffix or whatever for XDF-formatted images, as they're fundamentally different.

keirf, 2018-02-11:
I can auto-detect by size as it happens. I will wait for @claunia to get back to me with an XDF/HFE example. And yes of course now DSK and IMG suffix can be used interchangeably.

I'd be interested if @claunia has any luck with your OS/2 boot disks, depending how easy it is to setup that test (spare machine!).

claunia, 2018-02-11:
@keirf that disk2 is not XDF, XDF are 1800000 bytes approximately. I can send you the kryoflux of a real xdf where you want. https://github.com/claunia/CreateXDF creates empty .xdf image, no need to check by extension their sizes (both 5.25HD and 3.5HD) are quite unique. These empty images can be DIR A: with XDF.COM and OS/2, and can be written to real floppies with XDFCOPY. This is the format that is used as .xdf that's different from the real ondisk layout. XDFCOPY apparently support an ED variant but no one has found it (I have some disks that MAY be XDF-ED but don't have an ED drive right now). Just tell me where to send you the kryoflux dump please.

davide125, 2018-02-11:
Alright, I tried this again and realized I got things mixed up a bit. When it finishes reading disk 1 and prompts for disk 2, I hear half-second separated "clicks", but they actually come out of the system buzzer, not the gotek one (sorry for the confusion). The gotek buzzer is silent at this point, and the green led is blinking; when I switch it to disk 2, the green led goes solid and it makes a couple of "head seeking" noises, then I immediately get that error. Also worth noting, when switching to disk 1, I get read errors if I press ENTER immediately -- I need to wait about a second for it to go through.

keirf, 2018-02-15:
@claunia email me XDF (address in my Github profile) and you can discuss XDF support on new ticket #54

Invincing, 2018-10-18:
This behaviour sounds similar to the problems I've had with my Akai S1100, as described in issue https://github.com/keirf/FlashFloppy/issues/173

Try setting "pin02 = nc" in FF.CFG to see if it helps.

techana, 2019-02-08:
@kerif any news about the XDF support please? I'm trying to install OS/2 Warp 3 on a VM. Disk 2 and onwards cannot be read since they are in XDF format.

Here is .scp image for disk 2: https://www.dropbox.com/s/1f8xymdw63qbpt8/2.scp.zip?dl=0

The outer cylinder in XDF disks is divided into 9 sectors as in the standard disks (more info here: http://www.os2museum.com/wp/the-xdf-diskette-format/). This means DOS/Win will normally read any file(s) in this small portion of the disk.

The actual XDF data resides on the inner FAT, where each track is divided into 4 sectors only (8KB, 2KB, 1KB, and 512B respectively).

If you open the above .scp image in HxCFloppyEmulator software and click on Disk Browser, then you can see the PACKING.LST file, which reside in sector 4, side 0, track 0 on the outer FAT12 track. This means the image was created correctly using "scp_dump" and SCP card. But converting the .scp to any image readable by the emulator (such as .img) resulted in a corrupted image.

There are many tools circulating in the OS/2 community for dumbing XDF disks, either in OS/2 or Windows. Unfortunately, these tools require legacy FDD since USB are not supported. I noticed that the XDF images dumped by these tools don't have the outer FAT12 portion and can be mounted successful in many emulators, such as VMware and VirtualBox. Here is an example of such XDF images: 
ftp://osfree.org/upload/win1.dsk (more info here: https://www.os2world.com/forum/index.php/topic,1902.90.html)

Here is also an open source code for dealing with XDF images (including links to Win32 compilation): https://ecsoft2.org/xdflibsf-os2

keirf, 2019-04-04:
@claunia I had a look at your XDF/HFE images. I need to work out the correspondence between areas in the XDF/IMG file, and physical sectors in the HFE image. That's hard because the XDF images are empty. How did you convert the HFE to XDF/IMG (or vice versa)? Did it involve the black-box magic of XDFCOPY? Because I have seen no explanation or documentation on, for example, the sector numbering, ordering, or correspondence between IMG sectors and physical sectors...

What I really need is each 512-byte section in the XDF to be as unique as possible, at least once for each unique track layout (so let's say track 0 side 0, track 0 side 1, and any other single side of a single track), so that the correspondences can be mapped out by inspecting the equivalent HFE.

All the info about FATs is really of no interest. FlashFloppy doesn't work at the filesystem level -- it just wants to marshal raw image data into physical track sectors.

keirf, 2019-04-04:
Oh hang on, the OS/2 Disk 2 might be sufficient....

keirf, 2019-04-05:
XDF Supported in v2.2a. Please test and raise a new ticket for any issues.

closed 2019-04-05 by keirf