keirf/flashfloppy #29: Support for the HxC Mount

TzOk83, 2017-12-02:
Is there any chance for the HxC Mount support?

keirf, 2017-12-08:
I'm happy to work with someone to debug the problems.

TzOk83, 2017-12-08:
I may try to help, but have very little programming experience on ARM/non on MC68000. So, what do you want me to check for you? I have an Atari 1040STe with 2MB RAM, and switchable TOS v.1.06D/2.06D, and Megafile 60.

P.S.
There is a problem with ST disk images with over 80 tracks and/or over 10 sectors/track.

keirf, 2018-01-29:
Try using hxcmount and see how it fails. Hint: you may need to lie about the reported firmware version using FF.CFG option da-report-version = "V2.0.0.0" or somesuch.

If you can test and report, I can try some firmware fixes and send to you for further testing.

TzOk83, 2018-01-29:
HxCMount mounts the image, and files are accessible, is is possible to list text files, but it fails to load any PRG. I'm trying on 0.9.7a... what happened to synced track read option? ST image with 9 sectors/80 tracks reads again in 1:06 and but fmt+write is 1:20.

keirf, 2018-01-29:
Okay thanks, that behaviour has been reported before. I'll get back to you with a modified firmware to test.

keirf, 2018-02-27:
I haven't forgotten you. I will try to get you a firmware for test this week, so we can put this to bed for v1.0.

TzOk83, 2018-02-27:
As you've probably noticed, I've bought HxC license, and recently I was "playing" with it... but today my second GOTEK unit came from China, so I'm ready to test FF again.

P.S.
Jeff has criticized the idea of optimizing the performance based on a FastCopy III format, as this is quite far from the native Atari ST disk layout. So I'll prepare some file based benchmark tasks on both FC3 and native formatted disks.

keirf, 2018-02-28:
No problem, open another ticket if you have any performance issues relative to real floppy drives or other emulators on the various formats.

keirf, 2018-02-28:
Here is a build of the firmware hopefully fixed for HxC Mount. Please test!

[ff_hxcmount.zip](https://github.com/keirf/FlashFloppy/files/1767044/ff_hxcmount.zip)

keirf, 2018-03-01:
Here is an updated firmware for you to try with HxC Mount:

[ff_hxcmount_2.zip](https://github.com/keirf/FlashFloppy/files/1771283/ff_hxcmount_2.zip)

TzOk83, 2018-03-01:
It is slow (saving a simple text file takes "ages"), but it works... OLED display seems to be "frozen" only W blinks. All other is static:
hxcmount_100
008/014       T:99.0

keirf, 2018-03-01:
Is it not expected to be slow? HxC Mount requests a 64-sector track, which is 679,200 bitcells and takes 1.36 seconds for a complete revolution (ie. to spin back round to sector 0).

Requesting more than the default 9 sectors per track is an odd design decision, especially for a small file access which will be mostly random access in traversing dirents.

Anyway, let me know if FlashFloppy is especially slow. And how slow that is. And how slow it should be!

TzOk83, 2018-03-01:
Recently I'm very busy, and not have much time for testing. Read with HxC mount seems ok (it is slow as it is meant to be), but saving a test file containing test12345 from TEMPUS2 text editor took several seconds (I didn't measure it exactly, but I guess it was over 10 seconds).

keirf, 2018-03-01:
That's kinda weird then. I may have to set up an ST myself, or get you to grab serial logs from FlashFloppy on a future test run, if you can.

keirf, 2018-03-01:
In fact I urge you to open a ticket as then we can discuss the fix, which I think I know what it is ;)

closed 2018-03-01 by keirf