keirf/flashfloppy #35: IMG HD problems

Enzo1960, 2017-12-17:
IMG HD images (1.44 M) don't works well. When reading, sometimes give errors. When writing, only files of short lenght can be copied without errors.
Same files, in 720k images, can be copied well.
Same for format operation: they failed always, in HD mode, writing track 0.
Impossible to execute "SYS A:" command, for example, on already previously formatted HD 1.44 IMG.

keirf, 2017-12-20:
I think that is quite unexpected.... have you tried more than one USB drive?

Enzo1960, 2017-12-21:
Yes. I’m trying on an old Epson EL Plus (80286), but I think (slow) speed of computer is not relevant. However I’ll try on other PC and different usb sticks, snd let you know. Thanks!
Best regards and Merry Christmas too!

Inviato da iPhone

Enzo1960, 2017-12-21:
Tried now with other (s) usb sticks/ computers. Same results. Try yourself from plain dos “format b: /u /s” where b: is an empty (or however erasable) 1.44 img file you load from buttons. All goes well, but error at end of process.

Inviato da iPhone

Enzo1960, 2017-12-21:
On same hardware, trying “format b: /u /s /f:720” over a 720 img file all goes well. Errors, on 1.44M, happens when drive is at track 1.

Inviato da iPhone

keirf, 2017-12-21:
I'm not sure I actually have a PC with a floppy header... might have to pick one up.
I have heard that format doesn't work well. How about ordinary reads and writes?

Enzo1960, 2017-12-21:
On 720k format no problem, works like a charm. Problems always with 1.44M in write, sometimes while reading. I tried also on other Gotek hardware, and a lot of usb sticks.

keirf, 2017-12-21:
Ok thanks I will obtain an old PC which will allow me to investigate this issue more easily.

Enzo1960, 2017-12-21:
In my opinion problem isn't on formatting, because problems happens also on already generated on pc 1.44 img files, writing files on it, or trying to make images bootable with dos SYS command.
Perhaps interleave?

keirf, 2017-12-22:
A few other people have tested for me. One had success, another had problems as you do. I don't think it is interleave: PC HD floppies don't have interleave.

Enzo1960, 2017-12-22:
I tested also today on a Olivetti PCS86 (NEC V30) that supports only 720k floppies, and works as expected: on boot recognize only 720k images, giving error on 1.44 img.

Enzo1960, 2017-12-30:
Tries to do (now I'm in bed with fever and I can't try):
Format an 1.44M floppy with FreeDos instead MSDOS.
Format an 1.44M floppy under Linux, ext2 filesystem.

martinx72, 2018-01-04:
Same issue here, i test img/ima both got reading error at t49. any idea?
There is just only one big file in the image file.

keirf, 2018-01-04:
I don't have any idea as yet as I have not reproduced these issues, yet. I will soon lay my hands on a suitable PC mobo and then I will do my own testing. Should be in a week or two. :)

Enzo1960, 2018-01-04:
Further test on Olivetti PCS86 (NEC V30): I found jumper on mobo that allow controller to use 1.44M floppies instead of 720K, tried and also on this PC same results.

keirf, 2018-01-26:
Can someone re-test this with the following recent build?

https://github.com/keirf/FlashFloppy/files/1663928/ff_44_2.zip

Enzo1960, 2018-01-26:
Ok.I'll do now.

Enzo1960, 2018-01-26:
Tested now: seems it works! Format with /s parameter now works ok.
Further test soon.
Good work, thank you, Enzo.

2018-01-26 18:23 GMT+01:00, Vincenzo Femia <enzofemia@gmail.com>:

Enzo1960, 2018-01-26:
Other test: Game “Bombuzal”, that previously give an error if on virtual 1.44M, and runs ok on 720k, now runs also on 1.44M.
Command “sys b:”, now run well too.

Inviato da iPhone

keirf, 2018-01-26:
Thank you, I will close this issue as it will be fixed in v0.9.7a. Please re-open if the issue reoccurs after v0.9.7a.

closed 2018-01-26 by keirf