keirf/flashfloppy #36: Dual floppy support

Quantum-cross, 2017-12-19:
I see that in the documentation there is rumbling about being able to emulate two floppy drives at the same time. I am extremely excited at the prospect of this!

Would you mind if I open this issue to track the development and discussion of this feature?

keirf, 2017-12-20:
Yes that's fine. :)

pauldude12, 2018-01-19:
i would like to see a disk1 and disk2 chain txt for each games for adf and st formats or a windows program that merges 2 disk images in 1 larger disk image

pauldude12, 2018-01-19:
i know this was done on the c64 sd card to iec that used a switch and a text with images say 
batman side 1
barman side 2
and mounted as one image
also dreamloader for mmc64 had a batch converter that linked 4 d64 images into a dfi format but with this hardware its a stm mcu and depends on firmware space and holding drive routines into memory

pauldude12, 2018-01-19:
plus the atari only support single side disks 720kb so a 1.44mb image may not work as that what the floppy drive was we need a pc app that crunches disk from 1 and 2 and pack them to fix with side loaders

pauldude12, 2018-01-19:
sorry 800kb max a disk

kodak80, 2018-01-19:
Atari ST supports both single-side and double-sided disks.  The original STs were only single-sided disks but later modes were double-side.

pauldude12, 2018-01-19:
ok thanks i got st 520fm machine i got some hacked st games with trainers like batman already done from emu paradise as it was a 3.2 gb romset i have got all the game ever made

pauldude12, 2018-01-19:
sorry i mean batman disk 1 image and 2 image
into one st disk image

pauldude12, 2018-01-20:
@

pauldude12, 2018-01-20:
is there a way to convert 2 st disk in to a hfe format like extract the data from 1/2/3 disk and build a hfe image as one sorry im a c64 and sinclair so i am a newbie to this regards paul

v-retro, 2018-01-21:
I don’t think this is strictly a Flash Floppy (or even Gotek) related question, but in my experience, to achieve what you’re after requires:
- enough space on the disk (obviously)
- no copy protection that needs combining when merging
- the program’s ability to accept Disk 1 and 2 files from the same place (back in the day, I noted some programs looked for a file in the root directory to know which disk it was currently in the drive.  Other programs used to look for a particular disk label for this, which in this case probably won’t work).

I think your best bet might be to browse the various one of the various cracking group’s compacted disk compilations.  They’re basically gone through the hard work of addressing copy protection and space issues, such as Automation and Medway Boys.  These images work well with Flash Floppy too!

pauldude12, 2018-01-21:
thanks v retro ill check these out as ya copy protection was a issue till hackers started to decript codes and by pass it like block locks ect i though x copy was good on the amgia 1200 side back in the day

pauldude12, 2018-01-23:
every games needs a copy protection dll container to keep tracks and sectors in the same place as the orignal st amd adf for amgia as a ghost image with a update flash to suport drive lock codes

Sassa0, 2018-02-17:
Hi,
And first, keirf, thank you for your great firmware.

Back to the original enhancement topic : it should be even possible to use the four drive select signals in machines that have them, such as the Amiga.
In this case, on which pis of the STM whip would you connect them ? To stick to port A, PA6 (22) and PA13 (46) are unused at the moment if I am correct.

The reason I ask this is because I have worked on a new schematic, based on the one from Hervé Messinger, to integrate all the additions (oled screen, rotary encoder, mtron...), and correct the missing pulldown on BOOT0 and few parts specifications. I am interested in making my own PCB in order to integrate it nicely in my machines without having to cut their precious original case.

And this might also allow, for example, to replace the spi flash footprint by a sdcard header, in order to read the floppy images from there ;)

bni, 2018-10-03:
It would be awesome to have dual floppy, or even quad floppy support, when using the gotek with the Amiga, since you constantly have to insert the WB disk when using Workbench 1.3 at least.

I see that there is already some stuff about DF1, and being able to switch between DF0 and DF1 in the settings menu in AUTOBOOT.HFE. Is that related to this at all?

Couldn't it work like this?
AUTOBOOT.HFE has 4 independent lists of floppies that u can cycle, that match DF0, DF1, DF2, DF3. U can assign ADFs to these lists like normal.

Then when using the Amiga:
Long pressing one of the gotek buttons changes what list/drive is active for floppy swap.
Press up and down like normal to change floppies from the active list/drive.

DavidAPease, 2018-10-04:
Hi, Sassa0,

Just for your information, I very recently bought a Gotek unit (and am running FF, and have made several of the mods/enhancements on Kier's hardware mods page -- Thank You, Kier!).  

My Gotek board has an extra signal on the programming port, a fourth hole next to the JTCK-SWCLK pin (on Hervé's schematic).  This new hole/pin is connected to PA13, though as far as I can tell it is unused.   (I assume the Gotek folks must be planning some use for it.)  At least that could give you (or Kier) an easy way to add a signal to the CPU for the (presumably) latest units.

Sassa0, 2018-10-05:
Hello David.

Thanks for your message.
Actually I own two "standard" goteks. One came with the LCD, and it has this extra pin hole. The second one is newer, came with no LCD and no button, and also without this pin hole.

About the subject of the topic, when I was playing with the code, I understood that this feature would require substantial modifications in addition to the standard gotek hardware limitation.

Personnaly, my main project is a bit different : on the few machines that I own I want a solution to both enjoy my collection of real floppies and also floppy images from other sources on the usb stick. For that, I need to connect both the gotec and the original floppy drive inside the cases, an also be able to switch which one should be the bootable drive.
So the solution I imagined is a gotek board with a daisy chain header. Actually I have schematics and two versions of the board(s) fully routed on kicad (both having an internal IO part and an external HID part). Of course all the signals for up to 4 drives are routed. I also made a few tests on one of my goteks (removing few optional pullup resistors and routing additional signals out of the stm32). Modifying the code also helped me to understand few things and choose the right STM pins the the signal interrupts and rework my first schematics.

Actually, besides other spring and summer projects, what hold me is that Keif is doing a great job by adding new features, including for the Amiga which is my main target (drive ID, HD drive...). Thanks again for that ! I understood back then that I would not have the time to maintain and add all of that in a separate branch needed for the few additions of my board.

However I still plan to work on it, especially now that the nights are longer. The first goal is to publish "working" schematics and PCB files so that you people could use and improve it, and so that the few additions to the code that it will require would have a chance to land in the main branch ;)

Sassa

TzOk83, 2020-01-09:
Actually supporting multiple floppy drives does not require much additional coding. It is a matter of selection 1 of 2 (or 1 of 4) pre-selected images by DS0..3 signal, and maintaining current track for each image. If you have 2 physical drives, you can't access them simultaneously, the access is always interleaved.
On the hardware side you can bridge S0 with JB, and use PA2 to detect DS0, and put jumper across D1, to use PA0 line as DS1.

keirf, 2020-01-09:
@TzOk83 Are you offering?

vldmrrr, 2020-05-18:
I started playing with implementing this. Created two separate instances of drive structure to track the state of each drive independently, use jumper between S0 and JB for separate unit select from S1, use short press on both buttons to switch unit for slot selection, long press to eject. My changes seem to work with single unit, but crash with "Unexpected Exception" when switching unit, seems somewhere in or after timer_dma_init. Unfortunately, at the moment I completely do not understand the underlying logic of dma/timer working: does this need to to be separated for independent units. I assumed not, since only one unit is active at a time on the cable, but emulation logic evades me thus far. If there is any write up on how timer/dma is used any hint would be appreciated. Otherwise if you could find a time to take a look at WIP commit on 2disks branch in my fork repository and suggest further direction that would be even more appreciated. The branch is probably not even in compilable state but the intent of change should be quite clear. Thanks

keirf, 2020-05-18:
You try to get it working on a standard gotek and with simple three digit display? What a brutal UI haha! There are enhanced boards around with proper second drive select (unsupported at the moment of course) and it might make sense to require oled/lcd for this feature, I don't know. I also don't know if I'll support a hacked different second drive select pin for standard Goteks. 

I have a twodrive branch myself and it complements yours as it is more bottom up compared with your top down. Maybe it helps you. Of course it's nowhere near complete but shows how the low level logic can work.

richard-broadhurst, 2021-01-16:
I was wondering if anyone had been working on this and if there this would be supportable on "interface = shugart", possibly with an FF.CFG option or would the jumper be necessary?

wupeka, 2021-02-17:
Are there any plans to include it in master (or at least sync the twodrive branch to master?)

ukwebb, 2022-06-29:
hi folks - just wondering if there was any progression on this dual drive support?

coopzone-dc, 2023-02-03:
Has this got any further forward yet, it's a really good idea.

kivijakola, 2023-02-26:
I worked with this idea and got it passably working! (FW hack only)
Here is information about it: https://kivijakola.fi/projektit/2023/02/26/gotek-flashdisk-dual-drive-support/
And forked source code: https://github.com/kivijakola/flashfloppy

Quite lot of work is needed if want to use separate images for both drives and some selection mechanism for separate "drives" should be also designed.

ukwebb, 2023-02-26:
thats a great start @kivijakola :)  if nothing else, it at least it proves its possible. 
I'd say it does need to run from two separate image files and have the option to select drive0 image, and select drive1 image
but overall the drive physically cant use both at the same time, as the interface can only access one drive at one time, so its more a case of knowing if its a drive 1 select, use this image file, if its drive0select, then use the other image file

Dietrich-L, 2023-02-26:
I hope, the idea moves forward, because from a practical point of view it has some advantages. Especially it would allow to copy files from one image to another image on the same USB stick, which would be nice

Please go for it

Dietrich

kivijakola, 2023-03-12:
I have been preparing some updates for dual drive usage. It is now possible to store separate images for both drives in USB stick. DSKA* and DSKB*. 
[image]

There is big problems to get all data structures  to fit in the very limited 32kb ram. So, images must be strictly same format and size. It is very much impossible to get everything doubled.

But anyways, with some limitations it seems to be working quite smoothly:
[image]
I have only tested raw images now, but this should be working with other types also.

Lot of testing, clean up  and integration is needed. Hopefully someone else can continue from this :)

There is now updates in the fork: https://github.com/kivijakola/flashfloppy
And also project page: https://kivijakola.fi/projektit/2023/02/26/gotek-flashdisk-dual-drive-support/

keirf, 2023-03-12:
Nice. This would be better targeted at 435 based Goteks. Squeezing into 32kb ram is very hard.

iosoft, 2023-07-15:
SUGGESTION: the pendrive will have two separate partitions for two different emulated floppy-drives.

iosoft, 2023-07-15:
> I have been preparing some updates for dual drive usage. It is now possible to store separate images for both drives in USB stick. DSKA* and DSKB*. FF.CFG must configure with setting 'nav-mode = indexed' [image]
> 
> There is big problems to get all data structures to fit in the very limited 32kb ram. So, images must be strictly same format and size. It is very much impossible to get everything doubled.
> 
> But anyways, with some limitations it seems to be working quite smoothly: [image] I have only tested raw images now, but this should be working with other types also.
> 
> Lot of testing, clean up and integration is needed. Hopefully someone else can continue from this :)
> 
> There is now updates in the fork: https://github.com/kivijakola/flashfloppy And also project page: https://kivijakola.fi/projektit/2023/02/26/gotek-flashdisk-dual-drive-support/

Atleast make separate directory A and B

iosoft, 2023-11-09:
Any update on this feature?

tomchiverton, 2024-07-29:
This would be awesome to have either merged in, or at least added as an additional download option.

tomchiverton, 2024-08-08:
> > I have been preparing some updates for dual drive usage. It is now possible to store separate images for both drives in USB stick. DSKA* and DSKB*. FF.CFG must configure with setting 'nav-mode = indexed' 

Has anyone else been able to try this branch ? If so, on what hardware ? I'm a bit nervous of making a non-standard jumper change if it's going to maybe damage the hardware, for instance, because I have the more recent Artery based ones.