keirf/flashfloppy #37: Embroidery Machines

dRwH0, 2017-12-19:
Hello,

Will this be able to "recognize" formats used by embroidery machines such as Tajima/ZSK/etc ?

Thanks in advance,

keirf, 2017-12-20:
I suppose it could if I know what the geometry/layout is for these formats. Can you send me some examples?

dRwH0, 2017-12-20:
I have a SuperCard Pro, which I have successfully used to image/copy disks with it (IBM 1.44MB / Index Mode). I could give you an SCP file of an empty formatted disk (from a ZSK Machine) and one populated with files if it could help you.

keirf, 2018-01-08:
If there's no standard special image format for these machines, I would recommend try converting your SCP files to HFE using HxC tools. HFE should work no problem, you may need to set up the Gotek correctly though (eg. jumper JC if the machine expects a PC-interface drive rather than Shugart-interface, and set the jumper on S0 or S1 as appropriate for drive-select).

closed 2018-01-08 by keirf