keirf/flashfloppy #38: Annoying System Volume Information folder being displayed

LittleGem, 2017-12-20:
Hi,

Probably covered before (if so then please ignore). 

The System Volume Information folder is added every time a USB file system is touched by Windows. This folder redundantly appears in the FlashFloppy display. 

Perhaps it could be simply filtered out?

Sample code ... 

static bool_t native_dir_next(void)
{
    do {
        F_readdir(&fs->dp, &fs->fp);
        if (fs->fp.fname[0] == '\0')
            return FALSE;
        /* Skip dot files. */
        if (fs->fp.fname[0] == '.')
            continue;
        /* Skip sys vol directory */
        if ((fs->fp.fattrib & AM_DIR) &&  (strcmp(fs->fp.fname, "System Volume Information") == 0))
			continue;
        /* Allow folder navigation when LCD/OLED display is attached. */
        if ((fs->fp.fattrib & AM_DIR) && (display_mode == DM_LCD_1602)
            && ((cfg.depth != 0) || strcmp(fs->fp.fname, "FF")))
            break;
    } while (!image_valid(&fs->fp));
    return TRUE;
}

keirf, 2017-12-20:
Does the folder have AM_HID set in fs->fp.fattrib? Are you able to test that (have you successfully built the firmware)?

LittleGem, 2017-12-20:
Will check the attribute setting. Built and tested. Appears to work fine. Had the same issue with another project using fatfs-just filtered the directory from the listing in the same way. Could be made more generic by filtering hidden directories/files (already done? Haven’t checked the code too closely).

LittleGem, 2017-12-20:
Didn’t mean to close (new to Github)..

LittleGem, 2017-12-20:
Checking the attributes in windows gives SH (system, hidden) as expected.

keirf, 2017-12-21:
I will fix by filtering all hidden files & folders.

LittleGem, 2017-12-21:
Good call. Best solution.

commit 80e8d5da (2018-01-08): Ignore hidden files in native navigation mode. Fixes #38
closed 2018-01-08 by keirf