keirf/flashfloppy #39: Electronic Music Instrument Support

AkaiGuy, 2017-12-20:
Hello, 

Many Electronic Musical Instruments use Shugart; will FlashFloppy firmware work as generic Shugart drive?

It seems that by comparing these:

http://floppy.shugart.free.fr/page.php?page=materiels/atari/index.htm
http://floppy.shugart.free.fr/page.php?page=materiels/amiga/index.htm

to these:
 
http://floppy.shugart.free.fr/page.php?page=materiels/roland/index.htm
http://floppy.shugart.free.fr/page.php?page=materiels/akai/index.htm
http://floppy.shugart.free.fr/page.php?page=materiels/ensoniq/index.htm

There may be the possibility of using a Gotek drive with FlashFloppy firmware in these machines?

Thank you for all your research and work on this. It is much appreciated by all.

AkaiGuy

keirf, 2017-12-20:
What's really needed is a database/wiki of tested systems, and correct configuration for each. Many synths will already work fine with FF, if the Gotek and FF are correctly configured :)

I guess I could leave this ticket open to track the database :)

TzOk83, 2017-12-27:
Atari ST and Amiga also use Shugart (Seagate) interface and FF is compatible with them. Shugart has support for 4 floppies, with a common MOTOR ON signal, while PC floppy interface has support only for 2 FDDs with a separate MOTOR ON signal.

keirf, 2018-01-08:
Closing this one out.

closed 2018-01-08 by keirf