keirf/flashfloppy #42: Pasti image support

TzOk83, 2018-01-08:
It would be great if you could manage to add a support for Pasti disk images (if it is possible at all). This is a format and software for imaging copy-protected disks.
http://pasti.fxatari.com/

KodeMunkie, 2018-01-10:
+1 for stx support

keirf, 2018-01-11:
I think this would be hard to support directly in real-time on a Gotek. It would be fun to try though.

kodak80, 2018-01-11:
I would add a +1 to this request.  You can already use the HxC Emulator software tool to convert STX to HFEv2 or HFEv3 which works most of the time (depending on the copy protection used on the disk).  HFEv3 works even better if you have a raw dump to convert.  I have had great success converting Kryoflux and SCP dumps to HFEv3 and they work fine on my FlashFloppy Gotek.

pauldude12, 2018-01-19:
can i have a link to  HxC Emulator software tool please
regards 
paul

zarkers, 2019-04-14:
+1 for stx support

v-retro, 2019-04-17:
If there's a vote, +1 from me too.

Mi1ezDR, 2019-05-25:
+1 really, really want this!

keirf, 2019-06-18:
Took a closer look today, I don't think this can be done effectively in firmware. Even PC tools can't reliably convert all STX to HFE/HFE3: information is lost in the STX format in the cooking down of raw dumps into "records I should return in response to FDC commands" (eg. read sector N, read whole track, ...)

I don't really want to implement a feature where I will then get an endless stream of "STX of game X doesn't work" support requests because I couldn't do a 100%, or probably even a 90%, job.

What the ST could do with is a curated library of raw dumps, that users can take and convert as they need. I may also advise or help out in STX to HFE3 conversion, I feel a tool could be written that could get pretty close to 100% on that. There are smart people in the ST scene who could do this though, the formats are documented pretty well.

closed 2019-06-18 by keirf