keirf/flashfloppy #43: Amstrad CPC6128 Super Cauldron issue

Turk182es, 2018-01-18:
Hi,

I update with 0.96 firmware and still have Super Cauldron all versions doesn't work. I have blank screen. 
I have the same issue with more game with long loading.

Regards
Stephane

keirf, 2018-01-18:
Are you using DSK? Have you tried converting to HFE? Is the game an original or a cracked version with copy protection removed? (Sorry I'm not familiar with the CPC scene and whether distribution is generally of perfect images or of cracks).

Turk182es, 2018-01-18:
I'm using DSK not converted because since new firmware you can use DSK, that's why I onpen an issue just to let you know.
I try original version and hacked version with removed protection. Same issue.

Turk182es, 2018-01-18:
Just try and same issue with HFE, so it is an issue with the game, I have some other games with issue if you want to investigate. Most of them are game with Side A and Side B.

keirf, 2018-01-18:
Are they all double-sided, or are there some single-sided ones with issues?

Turk182es, 2018-01-18:
Super Cauldron is single side

keirf, 2018-01-18:
Ok, I had a theory involving being too slow to switch disk sides. Obviously not correct for a single-sided game! Can you send me the disk image(s), or a link to them? My email is in my Github profile page if you want to email.

keirf, 2018-01-18:
Actually I'm being silly, all the original titles will be single-sided at a time unless disk physically flipped, duh. :)

Turk182es, 2018-01-18:
I don’t have it right now. It I download it from CPC-Power.com

Turk182es, 2018-01-19:
Hi,

I sent you the 2 image disk that don't work.

Thanks for your support.

TheFrenchy75, 2018-01-21:
Hi,

The "Super Cauldron (UK,F,G) (1993) [Original].dsk" image works fine with the HxC firmware for gotek. The converted hfe file too.
I didn't tried them with FlashFloppy yet.

Turk182es, 2018-01-21:
The issue is with the Floppy USB.

keirf, 2018-01-21:
Okay that's interesting thank you. Perhaps you could try the following modified firmware and see if it works with DSK or HFE?

[ff_dsk1.zip](https://github.com/keirf/FlashFloppy/files/1650027/ff_dsk1.zip)

keirf, 2018-01-21:
By the way the cracked+trained version seems to have changed in the last day or so? I have a single-sided version with more than 40 tracks. But the new version seems to be a 40-track double-sided image, with sides A and B distributed in separate DSK files? Can I assume that neither works with FlashFloppy at the moment? ;)

keirf, 2018-01-22:
This firmware has patches on top of 0.9.6a, I simply didn't change the version number in this firmware build.

Turk182es, 2018-01-22:
Perfect! I will try it tonight!

Thanks

Turk182es, 2018-01-22:
Good news !

I don't know what do the trick, the new DSK or your Firmware patch but know it work like a charm.

Thanks for your support, you can close this ticket if you want.

keirf, 2018-01-22:
Probably it is the new image. Worth checking with latest unpatched firmware.

Turk182es, 2018-01-22:
What the firmware patch does exactly ?

keirf, 2018-01-23:
It alters DSK long-track handling. Rather than shrinking the bitcell size to fit the track in 200ms rotation time, it instead fixes the bitcell size and thus extends the rotation time. This is particularly relevant for the protected original DSK, which has protection tracks with overlapping sectors. The naive decode of these tracks does not overlap the sectors and hence we get a very long track (~50% longer than a normal track). If we reduce bitcell size accordingly then there is no way the host will sync to the read stream. Instead it is saner to just make the track take longer to read. The host is unlikely to care (eg. unlikely to notice that time between INDEX pulses is ~300ms rather than 200ms).

This patch is unlikely to have any effect on the cracked DSK images to be honest. So the fact the new cracked DSK works is probably because the crack is new or fixed in some way.

Turk182es, 2018-01-23:
Thank you for your help!

closed 2018-01-23 by keirf