keirf/flashfloppy #45: .MSA/.DIM image formats for Atari ST

TzOk83, 2018-01-25:
The .MSA can be considered as a standard image format in the pre-STeEM era. Native Atari-TOS utilities usually support this format. The .DIM format is very similar to the .IMG/.ST, but also is one of the "old" formats which were used on real Atari ST computers in '90s.
[FD_Image_File_Format.pdf](https://github.com/keirf/FlashFloppy/files/1664657/FD_Image_File_Format.pdf)

keirf, 2018-01-25:
Both of these can contain compressed blocks (even if it is very simple compression) making random access a bit of a pain. Not sure I will support these.

closed 2018-01-25 by keirf

Gunstick, 2021-05-19:
MSA is only supported for reading. read-write should not be a feature. Would be nice if people would not need to convert their images. Or implement an MSA to ST converter which will replace the file with an uncompressed version.