keirf/flashfloppy #46: Write issus on E-Mu ESI (O/S 2.10)

raybellis, 2018-01-29:
I'm trying to use v0.9.6a in an E-Mu ESI sampler running O/S v2.10.

By moving the jumper from S0 to S1 I've been able to get the drive recognised by the O/S, and also been able to format a virtual FDD (having first stored an empty 1440 kB file on my USB stick).

However when I try to save anything to the disk no progress is made, and the O/S reports "Floppy FIFO Overrun".    Please advise if there's anything else I can attempt to reconfigure, or known workarounds.

thanks.

raybellis, 2018-01-29:
I just noticed that 0.9.7a came out.  I've just upgraded to this and it does appear to work in initial tests.   Is that consistent with the changes in this version?

keirf, 2018-01-29:
Yes it is.

keirf, 2018-01-29:
One other question: are you using HFE images? And if so, is there a source "raw sector dump" image type that I could directly support instead (as I have for Akai synths in 0.9.7)?

raybellis, 2018-01-29:
I don't think I have the means to produce an HFE image yet (unless it's sufficient to just create an empty file on the USB stick named FOO.HFE and then have the ESI-32 "format" that).

What would be the advantage over using plain IMG files?

keirf, 2018-01-31:
Oh, IMG files are working for you "out of the box"? That's good. :) No advantage to using HFE files, carry on with IMG and be happy.

closed 2018-01-31 by keirf