keirf/flashfloppy #47: Disk read error when reading .hfe file on BBC Micro with 8271 controller

kgl2001, 2018-01-31:
Since upgrading to 0.9.7a I get a disk read error on my BBC Micro when trying to read an 80 track disk that has been converted to .hfe format. It appears to read track 0 ok because it reports the disk catalogue ok, but when trying to read anything more, I then get the disk error. Reverting back to 0.9.6a and it all works perfectly again.

keirf, 2018-02-01:
Do you get an informative error (eg. an error code)?

Can you upload the HFE image to this issue ticket?

Can you try putting the following config line in your FF.CFG file:
`track-change = realtime`

kgl2001, 2018-02-01:
Sorry. I should have provided that detail. The error I get is Drive fault 10 at 01/C1 ...although the 01/C1 part can change, depending upon what file I'm trying to load. I'm sure I will be able to find out what fault 10 means, but I'll need to do a bit of digging. This is generated from the DNFS file system ROM.

I'll try the FF.CFG suggestion later today.

[TimeAndMagikTrilogy.zip](https://github.com/keirf/FlashFloppy/files/1684472/TimeAndMagikTrilogy.zip)

kgl2001, 2018-02-01:
It's a bit odd. When I add the line to ff.cfg
track-change = realtime
it starts to work. But subsequently removing the line it still works???

keirf, 2018-02-01:
The line gets saved into Flash memory on the Gotek.

keirf, 2018-02-01:
Please have a good play around and make sure things are reliable. I'm now also suppressing INDEX pulses on drive seek, and that is apparently bad for BBC. Testing a disk image with files a long way from track 0 would be good :) If need be I can make the new behaviour auto- or manually- deselectable.

kgl2001, 2018-02-01:
Even with 'track-change = realtime' I'm still getting occasional errors:

[image]
File DCode3 starts at sector 0x060 (track 9, sector 6)
File Aint3 starts at sector 0x08E (track 0, sector 8)
File ACode3! starts at sector 0x206 (track 51, sector 8)

It's not consistent. Following the same sequence of loads and sometimes it works fine.

keirf, 2018-02-01:
That's expected really. I will prepare you a patched firmware to try tomorrow.

kgl2001, 2018-02-03:
Had an opportunity to test a bit further today. Firstly 0.9.6a works without any problem on both DFS 0.98 & DFS 1.20 (DNFS). With 0.9.7a I get different results with DFS 0.98 compared to DFS1.20. DFS 0.98 seems to be a bit more reliable, but both give drive errors - particularly when trying to load files from tracks around 40 - 50. DFS1.20 consistently gives a drive error when I try to load GData3! from the file I posted earlier.

keirf, 2018-02-03:
Can you please try the following firmware with the following lines added to FF.CFG:
```
track-change = realtime
index-during-seek = yes
```

This will probably become a requirement for reliable operation on BBC systems with 8271 disc controller and DFS <= v1.20.

[ff_47_1.zip](https://github.com/keirf/FlashFloppy/files/1691863/ff_47_1.zip)

kgl2001, 2018-02-03:
I've reflashed with the FF_Gotek-v0.9.7a.upd file from ff_47_1, and updated the ff.cfg file with the two changes suggested, but unfortunately that doesn't seem to have made any difference. I'm still getting drive error 10 when I try to load the GData3! file.

keirf, 2018-02-03:
Apologies, there was a logic bug in the previous firmware. Please try this one instead :)

[ff_47_2.zip](https://github.com/keirf/FlashFloppy/files/1691989/ff_47_2.zip)

kgl2001, 2018-02-03:
That's working much better. I'm not getting any errors now.

Thanks.

commit b7ffe926 (2018-02-03): New config option "index-during-seek = yes|no*".

Allows reliable operation on older host systems which may depend on
original floppy drive behaviour, consistent index pulses while the
disk is spinning (never suppressed eg. during seek operations).

Reliable operation on BBC micro requires following FF.CFG lines:
index-during-seek = yes
track-change = realtime

Fixes #47
closed 2018-02-03 by keirf