keirf/flashfloppy #48: Feature request: Navigating out of folders

kgl2001, 2018-02-03:
I have set autoselect-folder-secs = 0 so I need to press a button to navigate in to a folder. Current behaviour when navigating back out of that folder is step back to the first file in the parent directory. Would it be possible to change this behaviour, so that you just return back to parent folder instead? That would be helpful with browsing when there are many folders on the USB.

I recognise that it may not work quite as easily when autoselect-folder-sec is set anything other than 0.

keirf, 2018-02-03:
So you would like to return back to the subfolder's position within the parent folder? The thinking on current behaviour is it makes it quick to return N levels up, as ".." is the default selection at each step. But your behaviour could make sense too.

kgl2001, 2018-02-03:
Yes. That's exactly what I was trying to say. You said it more eloquently. Perhaps make it a configuration option so you can have either behaviour?

commit 35065608 (2018-02-23): navigation: Return to subfolder position in parent folder when selecting "..".
Fixes #48
closed 2018-02-23 by keirf