keirf/flashfloppy #49: Linux mount errors

szg0000, 2018-02-08:
Hi!

I tried mount the Gotek floppy drive with FlashFloppy 0.97. There are error messages in the kernel log:

```
[ 9588.700023] floppy driver state
[ 9588.700024] -------------------
[ 9588.700035] now=2322176 last interrupt=2321283 diff=893 last called handler=recal_interrupt [floppy]
[ 9588.700036] timeout_message=floppy start
[ 9588.700037] last output bytes:
[ 9588.700040]  0 90 2321283
[ 9588.700041] 1a 90 2321283
[ 9588.700042]  0 90 2321283
[ 9588.700044]  3 80 2321283
[ 9588.700045] c1 90 2321283
[ 9588.700047] 10 90 2321283
[ 9588.700048]  7 80 2321283
[ 9588.700049]  0 90 2321283
[ 9588.700051]  8 81 2321283
[ 9588.700052]  4 80 2321283
[ 9588.700054]  0 90 2321283
[ 9588.700055] e6 80 2321372
[ 9588.700056]  0 90 2321372
[ 9588.700058]  0 90 2321372
[ 9588.700059]  0 90 2321372
[ 9588.700060]  1 90 2321372
[ 9588.700062]  2 90 2321372
[ 9588.700063] 12 90 2321372
[ 9588.700065] 1b 90 2321372
[ 9588.700066] ff 90 2321372
[ 9588.700067] last result at 2321283
[ 9588.700068] last redo_fd_request at 2321277
[ 9588.700071] 78                                               x
[ 9588.700086] status=50
[ 9588.700088] fdc_busy=1
[ 9588.700090] do_floppy=main_command_interrupt [floppy]
[ 9588.700092] delayed work.function=f84322a0 expires=1
[ 9588.700094] cont=f843bd34
[ 9588.700095] current_req=f5a86000
[ 9588.700096] command_status=-1

[ 9588.700102] floppy0: floppy timeout called
[ 9588.700108] blk_update_request: I/O error, dev fd0, sector 0
[ 9588.700198] floppy: error -5 while reading block 0
```

keirf, 2018-02-08:
Any success with earlier FF versions, or is this your first test with Linux?

Have you any success on this system with another OS?

Finally, assuming this is a PC system, have you jumpered S1 and JC (and no other jumpers at rear of Gotek)?

szg0000, 2018-02-08:
This is the first time with FF & Linux.
I didn't use the jumpers. FF is different than the GOTEK firmware. Is the FF jumpering the same as GOTEK firmware jumpering?

keirf, 2018-02-08:
Make sure you have jumpers at locations S1 and JC, and no other locations.

szg0000, 2018-02-08:
Thanks. With the new jumpering I have less error messages.
Now I have this:

```
[  252.616074] print_req_error: I/O error, dev fd0, sector 0
[  252.616295] floppy: error 10 while reading block 0
[  252.644067] print_req_error: I/O error, dev fd0, sector 0
[  252.644287] floppy: error 10 while reading block 0
```
I just pushed a 16G FAT32 pendrive into the GOTEK drive, and tried to mount with
`mount /dev/fd0 /mnt`

keirf, 2018-02-08:
Probably now a timeout error from the FDC (saw two index pulses but no valid sector to read) rather than the original multi-second timeout in the Linux kernel.

Have you tried an original Gotek or real floppy drive in this exact setup?

What files do you have on the USB stick? What does the Gotek display say?

TzOk83, 2018-02-08:
FlashFloppy is not like Gotek. It uses regular disk image files on the pendrive, rather than the pendrive partitions. So you have to create an empty floppy disk image in any supported format (preferably .IMG or .HFE) and put it on your pendrive, in indexed mode you may have as many as 1000 disk image files on the pendrive, named DSKA0000.IMG - DSKA0999.IMG. To use Indexed mode you have to put a valid FF.CFG configured for indexed mode in a root folder. Without it you will be in Native mode, where image files can have any name, but they will be assigned to slots according to their order in the FAT table (it is much better to have an OLED/LCD connected to Gotek, to see the file names in Native mode).

Please see the Wiki about it: https://github.com/keirf/FlashFloppy/wiki/Operating-Modes

szg0000, 2018-02-09:
What are the supported file systems to store the IMG files?
Is FF support 2,88M floppy images or bigger?

keirf, 2018-02-09:
You must format the USB drive as FAT32. On Linux mkfs.vfat should do the right thing for a 16G stick.

2.88M is the only geometry of ED image that is currently "supported". But even that is untested and FlashFloppy may well fail to achieve that data rate without some more optimisation.

I recommend you get 1.44M HD images working, and then try 2.88M ED. If the latter fails, the report belongs in a new issue ticket.

keirf, 2018-02-10:
I assume this is resolved. Please reopen if there are ongoing issues with 1.44M images.

closed 2018-02-10 by keirf