keirf/flashfloppy #51: Support both for PC and Amiga?

mesmariusz, 2018-02-08:
Hi.

Does it mean that FlashFloppy firmware for Gotek supports PC images. And finally Gotek with FlashFloppy + pendrive including several .ima / .img (PC) and several .adf (Amiga) can be connected both for PC and Amiga without reflashing a firmware to switch between PC / Amiga drive emulator?

[image]

keirf, 2018-02-09:
Yes, that is correct.

Note that connecting to a PC requires jumpering JC and S1, while connecting to an Amiga requires jumpering S0. See here: https://github.com/keirf/FlashFloppy/wiki/Host-Platforms

closed 2018-02-09 by keirf

v-retro, 2018-02-09:
My Atari Falcon and Atari Mega ST 4 each have 286 emulation hardware.  I was hoping that the Flash Floppy drive could use both .ST and .IMG images at the same time so I don't have to convert to a uniform disk format as simply use them in their native format.
I also have Mac emulation hardware for the ST but I know the ST's disk controller couldn't handle it (even if the Gotek/Flash Floppy could)

keirf, 2018-02-09:
Oh ST / IMG at the same time on Mega ST should be fine. You will be physically going through the ST's drive interface and physically configure the Gotek for that (which probably means jumper S0 only).

mesmariusz, 2018-03-30:
Hi.

> Note that connecting to a PC requires jumpering JC and S1, while
connecting to an Amiga requires jumpering S0.
> See here: https://github.com/keirf/FlashFloppy/wiki/Host-Platforms

Only hardware jumper switching is enough? What about FF.CFG - is this
config file mandatory or optional?
Should be created as a single one on an USB drive? Or this should act as a
pair for every .ima, .adf etc?

Other is there any difference between IBM PC (.ima files) and:

E-mu ESI-32
Requires jumpers at S0 and JC. Works with 1.44MB (HD) IMG files.

Why for E-mu ESI-32 S0 and JC and for IBM PC S1 and JC (means different
hardware, but the same .ima file format?)

Is there somewvere an easy explanation for all J5, JA, JC, JB, S0, S1, M0.

Sorry for this huge number of hestions. I appreciate all your work and
answers.

Best regards
Mariusz Ciszewski