keirf/flashfloppy #52: Amstrad 6128 protected games speedlock/hexagon protections

Greatfunky, 2018-02-10:
Hi, 

I noticed Amstrad games using latest disc protection from 1989 (not sure about earliers) are not working :  speedlock  or hexagon protections.

Even converted to hfe , they do not work, i tried also the patched firmware 0.9.6a from  another thread issue without success .

At first i tested:
Double Dragon (UK) (128K) (1988) [Virgin Mastertronic] [Original].dsk  (the cracked version is working fine)
[http://www.cpc-power.com/index.php?page=detail&onglet=dumps&num=750](url)
Then some others from the list :
[http://www.cpc-power.com/index.php?page=protection&lenom=Speedlock D7 - 1989](url)

and Final Fight [http://www.cpc-power.com/index.php?page=detail&onglet=dumps&num=875](url) with hexagon disc protection , cracked version is working fine.

Thanks.

keirf, 2018-02-10:
Original protected images will be generally unlikely to work. I have had some advice on improvements to make. I may get round to these in time, or not. Or perhaps someone in the CPC scene with DSK knowledge will contribute patches ;)

closed 2018-02-10 by keirf

Greatfunky, 2018-02-11:
I hope that will be supported in the future , thanks for the info !