keirf/flashfloppy #53: The latest firmware makes DSK images read-only

Wierzbowsky, 2018-02-11:
The latest 0.9.8a firmware doesn't allow to write into DSK images. They are read-only no matter what settings are used. Tried on MSX system with USB2 and USB3 sticks, FAT32-formatted, with default and custom FF.CFG, in both IBMPC and Shugart modes.

keirf, 2018-02-11:
I've generalised the title since this is not the only report I've had for v0.9.8a. Someone else reported problems on Sam Coupe (CPCDSK images) and there is this report now on MSX (DSK which are actually raw sector images I believe). Also someone reported autoboot "LBA write fail" on Atari ST.

So I will use this issue ticket to track the whole lot. I may also post updated firmwares to test here.

Meanwhile most users are recommended to stick with 0.9.7a for the time being.

keirf, 2018-02-12:
Could you try the following firmware? It reverts some patches from v0.9.8a, so will help confirm the bug really is in the major write-pipelining patch (or not).

[ff_53_1.zip](https://github.com/keirf/FlashFloppy/files/1715839/ff_53_1.zip)

jmeuwiss, 2018-02-14:
Hi keirf, my GOTEK had official 0.9.7a flashed, which worked OK on MSX dsk images (indeed these are 360KB or 720KB raw sector images).
Next I flashed 0.9.8a and I could reproduce the problem. Saving new files or updating files does not work (updates are lost and new files are not created).
Next I flashed ff_53_1: problem is gone. I can update files and create new files.

Hope this helps and thanks for also looking into the problems MSX users are experiencing!

kiwatiger, 2018-02-15:
Hi, i come to report that yeah, in ST mode is also read only, i can write and create files on my floppy but changes are not saved to the file, flashed ff_53_1 and resolved the issue.

Thanks for the project!

keirf, 2018-02-16:
Thanks for the reports. That's enough to confirm within which patch the bug must lie.

keirf, 2018-02-19:
Reproduced this issue on Atari ST. I will investigate further.

commit 8c39670b (2018-02-19): floppy: Fix corrupted buffering of last mfm word of a write.
Fixes #53
closed 2018-02-19 by keirf