keirf/flashfloppy #55: Thinner Font

MarcMuc, 2018-02-18:
Hi keirf!

First of all: wonderful work! Thank you very much for this!

But... :D

Is it possible to use a thinner font for OLED-Displays? I have the comparison to the HxC-Version and the font used there is for my opinion more elegant and better readable!
I attached a picture of it. 
[image]

Thanks a lot and keep on rockin'!

Marc

keirf, 2018-02-18:
Yes agreed the fonts could be better. What's that one -- probably 6 px wide, maybe 11 or 12 tall? If we can find a suitable nice bitmap font of that dimension then I will happily trial it.

keirf, 2018-02-18:
Orp Medium? https://github.com/MicahElliott/Orp-Font

keirf, 2018-02-18:
Tamsyn 6x12r: http://www.fial.com/~scott/tamsyn-font/

keirf, 2018-02-18:
If we can find an elegant small font I would be happy to move everyone over to it, and make the inter-character spacing configurable.

MarcMuc, 2018-02-18:
I would vote for Leggie. Tamsyn is also a good choice. Orp has serifs which makes it hard to read in that tiny size.

wiktor-b, 2018-02-19:
Hi, I'm the author of leggie. First of all I'm thrilled you're considering using my font in your project!

The problem with using a 12px tall font is you'll have a lot of "wasted" interline space, considering you have two-line 32px high displays. it would make more sense to design a slightly taller 6x16 font - it will be more readable from a distance, and you'll still get the extra 5⅓ columns.

You could also use a 6x10 font which would give you an extra line, or one that's 5px wide but that might come out too narrow.

The font in your picture is the classic `-misc-fixed-*`, 6x12 or 6x13 https://www.cl.cam.ac.uk/~mgk25/ucs-fonts.html .

I'd be very happy to contribute a/some font(s). It's a small time hobby of mine. I rarely have the chance to put any out, but I've spent a lot of time studying small bitmap fonts and I like to think I know how to design them :-)

keirf, 2018-02-20:
I will work on extracting BDF/FONT files to the required firmware format and then we can compare some of these fonts for readability on the OLED screen. Thanks for your offer of help @wiktor-b !

keirf, 2018-02-21:
Leggie:
[image]

Tamsyn:
[image]

keirf, 2018-02-21:
Apologies for the poor photos. I suspect the HxC font is 6x13 as it is noticeably taller than these two examples. Of these two I feel that Tamsyn is a bit less "stylised", particularly the numerals. Perhaps a taller font would be preferable however.

keirf, 2018-02-21:
The classic 6x13 font, clearly same as HxC, and clear winner here. Unsurprising really as it has an extra row of height to play with to make a clearer more balanced font:
[image]

keirf, 2018-02-21:
I will definitely replace the existing stylised 7x16 font with the classic 6x13 font. I may also get rid of 8x16 as 6x13 is clear enough. But maybe not immediately, and I will phase it out...

I am also open to comments/improvements on the 6x13 font :)

closed 2018-02-21 by keirf

keirf, 2018-02-21:
Updated firmware demonstrating new font (still only 16 char per row, for now):
[ff-font.zip](https://github.com/keirf/FlashFloppy/files/1744654/ff-font.zip)

commit d3ec20b9 (2018-02-21): oled: Replace 7x16 font with classic 6x13 font. Make it default, but
keep the 8x16 font around (for now).
Fixes #55