keirf/flashfloppy #59: Dual button only config Gotek navigation issues

salacpavel, 2018-02-24:
With 0.9.11.a firmware, it is now somewhat more difficult to use dual button-only configured Goteks. Basically with "twobutton-action = eject" in config file, when I press both buttons on a file, I randomly end up with inserted image, ejected image, or up/down navigation event. Thus I have to repeat dual button click few more times untill it does requested action. 
With 0.9.9a everything seems to work all right.

TzOk83, 2018-02-24:
On what platform you use your Gotek/FF that you have to "eject" an image? I've always wondered what for the "eject" function is :/

salacpavel, 2018-02-24:
Sometimes, you just need to "eject" floppy to interrupt whatever you are reading/writing from it, on any platform.

keirf, 2018-02-26:
@salacpavel I've been unable to reproduce this problem. I gave a quick test of v0.9.11a and v0.9.9a on both 7-seg and OLED displays, eject/insert with two buttons seemed to work same on both versions with both display types.

So I need more details on your setup:
1. What display do you use? (7-seg LED, LCD, OLED)
2. Please attach your full FF.CFG file
3. Please ensure any old Flashed config is wiped (hold both buttons for 2s with no USB stick inserted, will see RST or Reset message on display).

keirf, 2018-02-27:
Closing as can't repro. Please reopen if it occurs with latest v0.9.12a and include the extra details I list in the previous comment.

closed 2018-02-27 by keirf