keirf/flashfloppy #61: 600RPM

BABNIT, 2018-02-26:
Hello Sir,
Kindly help to Flash floppy for 600 rpm FDD

keirf, 2018-02-26:
Example images for Sony 600rpm drive:
[600rpm.zip](https://github.com/keirf/FlashFloppy/files/1758152/600rpm.zip)

keirf, 2018-02-26:
A few things needed here:
1. Remove hardcoded 300RPM assumptions and allow to be configured by image parser or FF.CFG.
2. Detect RPM if possible in HFE parser
3. Possibly need HFE HD rate optimisations, as data rate is same as a 300RPM HD disk which is unreliable at the moment due to lack of FlashFloppy buffer cache.

keirf, 2018-02-27:
Please try the attached firmware. No promises, especially as FlashFloppy currently struggles to serve up HFE images at high-density rate (which is what 600rpm DD requires).

[ff_61_1.zip](https://github.com/keirf/FlashFloppy/files/1763387/ff_61_1.zip)

BABNIT, 2018-02-28:
Thanks ,ok I will update this firmware and revert.

Regards
Nitin Babbar
961/8 ground floor
Govindpuri,kalkaji
New Delhi 110019
PH-26460260
MOB 9312274602

BABNIT, 2018-02-28:
Great ,its working ,Lots of thanks.

commit 98c99cca (2018-02-28): hfe: Determine rotation rate from HFE data-rate and track length.
Fixes #61
closed 2018-02-28 by keirf