keirf/flashfloppy #62: HxC Mount - write performance

TzOk83, 2018-03-01:
Read with HxC mount seems ok. Read is slow but acceptable, yet saving a test file containing a text 'test12345' from TEMPUS2 text editor took several seconds (I didn't measure it exactly, but I guess it was over 10 seconds), which is way too slow.

keirf, 2018-03-01:
Direct Access mode is currently resetting track offset to zero after each sector write. ie. read stream restarts at sector zero, rather than continuing straight on to sector N+1. That is probably screwing write performance here. I will cook up a fix and post it here for test, in the next few days.

keirf, 2018-03-02:
Here is a firmware for you to try:

[ff_62_1.zip](https://github.com/keirf/FlashFloppy/files/1775014/ff_62_1.zip)

TzOk83, 2018-03-05:
With recent firmware, writing 16B file takes about 12 seconds...

keirf, 2018-03-05:
Which firmware is that?

TzOk83, 2018-03-05:
This was on the official v0.9.14a... not the one from this topic, but I believe you have included this patch in v0.9.14a, didn't you?

keirf, 2018-03-05:
Yes indeed I did. Hmmmm. Can you also try with the firmware that improved ST timings? That's: https://github.com/keirf/FlashFloppy/files/1782262/ff_44_1.zip

That also happens to include fixes for index timing on very long tracks. Could be relevant here. If it doesn't help I'll have to try other things, like adding interleave to the sectors.

TzOk83, 2018-03-05:
I really have to make another cable to connect the second GOTEK unit... I'm constantly swapping them now.

keirf, 2018-03-06:
Hxcmount only supports Gotek as Drive A, right? Reading the source code that does appear to be hardcoded in the FDC access layer.

Having read the hxcmount code I think perhaps the following may help:

[ff_62_2.zip](https://github.com/keirf/FlashFloppy/files/1784003/ff_62_2.zip)

TzOk83, 2018-03-15:
I'm reopening just to confirm that now it works even faster than on HxC. Copying above mentioned file takes about 4,5 sec. and mounting image is almost instantaneous compared with HxC (always was).

keirf, 2018-03-15:
Excellent, pleased to hear it. I may take the hxcmount approach on Amiga too. I would like to make a block device driver there so that the whole USB stick FAT filesystem can then be mounted onto Workbench :)

Squirrel61, 2018-03-15:
WOW! That would be huge!