keirf/flashfloppy #63: How to use the images I have on the USB stick?

cph30075, 2018-03-02:
For the life of me I just can't figure out how to make/use images on the USB stick. I have multiple images that end in .imd or .td0 (those are images, right?) for my old Kaypro 10 I'm trying to restore, but when placing them on the stick and plugging it in for testing (It's only on power ATM, don't want to plug it into the Kaypro for good until I figure this out) it gives me an E34 error code. The Wiki says there's no entries to navigate, so I feel kinda dumb because there's definitely a way to solve this, but I don't know how. Do I have to convert them? wat do? 

Also, I'm a big noob to Github so apologies to the Developer (awesome firmware, I'm just a big dummy :P) or anyone else if I posted in the wrong place. Thanks for any help.

PeterBBCUser, 2018-03-02:
you need to convert them to .hfe files to work with the gotek using HxCFloppyEmulator.

keirf, 2018-03-02:
Yes need conversion first. IMD is possible to directly support in future. TD0 images are compressed so they won't be.

closed 2018-03-02 by keirf

cph30075, 2018-03-02:
Thanks, worked perfectly. Thanks for the neat firmware, by the way, @keirf