keirf/flashfloppy #64: repeating transmission

Mouriii, 2018-03-02:
Hello people i have the gotek on my yamaha A-100 psr(keyboard arrenger) i put firmware v0.9.13a and OLED screen, then i wrote in usb flash memory my images DISKA0000.HFE~DISKA0020.HFE with HXCSDFE.CFG(indexed_mode) then my yamaha's screen show for one second data from image then lost everything and is repeating loading without end. on OLED show T:0 and T:1 that is change every sec. what is the solve of my problem?? thanks a priori

keirf, 2018-03-03:
Attach a zipped HFE image to the ticket here. If you have the images in any other format (e.g. *.IMG files) then include one of those in the zip too.

Mouriii, 2018-03-03:
[DSKA0000.zip](https://github.com/keirf/FlashFloppy/files/1777362/DSKA0000.zip)
i tried to put all of types (hfe,hfeV2,hfeV3,img etc.) but only the hfev3 is reading, otherwise i have problems(say floppy not exist).

keirf, 2018-03-03:
That is a standard 1.44MB DOS disk. You should fine IMG file works absolutely fine. Did you try putting a jumper at location JC on the rear of the Gotek? That may be required. Also is the select jumper at location S0 or S1 (just checking, so I can document this setup in the wiki).

Mouriii, 2018-03-03:
You are Pro.. i put jumper on JC(i dont knew about that option) location and its working with existing image (HFEv3). i tried and without OLED and working perfectly.Thanks a lot

keirf, 2018-03-03:
I don't recommend to use HFEv3 except for certain kinds of copy-protected disk. Convert to a plain old HFE instead. Or better still, use 1.44MB IMG files with FlashFloppy directly.

closed 2018-03-03 by keirf

Mouriii, 2018-03-03:
Mr Keirf i have two problems. the first one is, when i try to save something music style(STY) from keyboard it say not floppy but when i want to save some sound(VCE)then it save it, also if i want to make a folder it make it.I made and format disc from keyboard and make it. i made as you said me, image of 1.44MB IMG. And the second one is, i tried to flash with TTL usbflasher(pl2303) but it say unrecognized device.

keirf, 2018-03-03:
pl2303 devices are often clones and you need a hacked driver on Windows which doesn't blacklist clone device ids. http://www.totalcardiagnostics.com/support/Knowledgebase/Article/View/92/20/prolific-usb-to-serial-fix-official-solution-to-code-10-error

So you are now using IMG files and are having trouble saving to those? Or are you trying to write to HFE?

tenuke, 2018-03-03:
'unrecognized device' problem is usually fixed just by resetting the
gotek. Gotek
pinout picture
<https://www.ikod.se/wp-content/uploads/2015/01/PinOutGOTEK.jpg>
have you tried that already?

Mouriii, 2018-03-03:
well i have installed the right driver of pl2303(clone) i tested it with software and working perfectly alone(RX to TX then i saw the transmit data to receive).i made reset as you say tenuke but same result.i am using IMG file, i tried with HFEV3 but same problem, i can save everything except Style/Rythm(STY).

keirf, 2018-03-03:
By the way once you have flashed a Gotek once you know you can update it via USB stick?

keirf, 2018-03-03:
As for the save problem... what are the sizes of typical STY files, and VCE files?

Mouriii, 2018-03-03:
yes i flashed with jtag the gotek just i had a question about TTL its not important. STY are 10KB-50KB and VCE(sounds) 1KB. i have much emty place in image to save up of 20 styles.

keirf, 2018-03-03:
So you are having trouble saving files bigger than a few sectors... What is the exact error when you try to save a STY?

Mouriii, 2018-03-03:
when i order to paste STY in image from keyboard then say "now executing" and i show like writing sometihng but after 10sec it say "no floppy disk", when i put " ok" then its reload as normal(before) the image.

keirf, 2018-03-03:
There will be a new release tomorrow, so try again with that. If it still fails feel free to open a new ticket, especially if you are able to collect debug logs from the working Gotek (that would mean connecting up the USB-TTL serial adapter while in situ in the keyboard and collecting serial logs via a host PC or say Raspberry Pi).