keirf/flashfloppy #65: Hardware enhancement using custom board

Sassa0, 2018-03-03:
I have a CPC464, Amiga 500, CPC6128, Amiga 1200 and Atari 520STF and I have been using a Gotek drive with Flashfloppy on them. Thanks ;)

I made  the hardware modifications to the Gotek, but I am still unsatisfied by its hardware.
Since I have few ideas, I came up with a new draft schematic inspired by the one from Hervé Messinger, including the available modifications, and few more.

Years ago when I bought my Gotek I contacted HXC to see it was possible to add those kind of functionalities based on their firmware, but it was not made possible back then. Well I prefer open things ;)

By the way, I am not a company. I don't plan to sell it like the Gotek or the HXC, I just want to make few PCBs for myself and friends, and share it with the community for anyone to make its own board, and so that these functions would be added to firmwares like FlashFloppy.

Before sharing the schematic and encouraging people to make their boards, I'd like to to agree on the pins to use for these new functionalities.

I came up that having an internal solution is better since:
- The Atari and CPC do not provide a power source in their external ports,
- To make the drive bootable (and compatible with all games), you need to open the case anyway: on the Atari to disconnect the internal drive, and on the 6128 and Amiga because DS0/_SEL0 is not exposed on the external connector,
- When inside the case, it is possible to pick up all the signals required to emulate up to the 4 possible drives on the Amiga,
- It make it possible to daisy chain the mechanical FDD, and dynamically set it on any of the 4 drive select lines, or even disable it, 
- The Amiga and ST external connectors are rare and expensive.

Based on that, our new board should:
- connect on the ribbon cable to the computer
- have a ribbon connector to the internal drive
- have the logic to send the identification of the 4 drives to the Amiga : ideally do it inside the STM32
- have connectors to the additional signals required : DS1, DS2, DS3, MTRX (on the Amiga, the MTRON signal from the ribbon will not be usable), DRES
- generate MTR and DS signal for the mechanical FDD : ideally do it inside the STM32, and have a menu to set it up
- have a connector to an external board with all the user interface. They should be connected together using a flat cable through the vents so that NO case modification is required (flat flex so it can be easily installed and removed).
- on the external board, have:
    the USB port,
    a SD port, since I read that it was available on old versions of FlashFloppy (connected to the SPI port that is used to the SPI flash footprint on the Gotek),
    the left/right buttons or the rotary encoder, depending on case design : make all the connexions available on the flat flex,
    the I²C oled screen,
- the buzzer might be inside the computer case for better resonance, but I may be wrong. (Actually I have used only piezo discs directly on JB, do you think it is worth using a magnetic one ?)
- the USB port power should be protected with the STMPS2141, because our old computers deserves that.

So here are the additional pins that I propose, I don't know much about the STM32 so I am not sure it is the best placement:

    DS2 input = PA6
    DS3 input = PA13

    For the Amiga signals:
    DRES = PC0
    MTRX = PC1

    For the mechanical floppy:
    MTR = PC2
    DS = PC3

    For the SD card:
    CD = SPI_CS = PB12
    CMD = SPI_DI = PB15
    CLK = SPI_CLK = PB13
    DAT0 = SPI_D0 = PB14
    WP = PC4
    DET = PC5

    Not used yet but why not connect it:
    Floppy ribbon pin 4 (INUSE input) = PB2

Other thoughts:
- The schematic from HM is missing the 4.7K pull down resistor on BOOT0.
- I have completed the schematic with few missing or incorrect part references.
- Will the timings be OK to control the mechanical floppy DS and MTR through the STM32 ? Beware that the MTR need to be buffered by a flip flop (see external drives schematic using DRES, SEL and MTRX)
- The Amiga drive identification mechanism (also in external drive schematic using ready, SEL, MTRX) should also be reproduced in software by the STM32
- Since we then need 8 outputs, I am wondering about simplifying the hardware by using something like a ULN2803 instead of the 74HC04 + NPN transistors + resistors (few parts to assemble, and even possible to use a though hole version of the chip). However all the output signals would need to be reversed in the software (possible ?), and the timings will also need to be tested (75ns for the 74HC04 + 250ns for the MMBT2222 transistor vs 1microsec according to the ULN2803A datasheet : will it be enough, especially for the data read line ?).

keirf, 2018-03-03:
There are a few extensions already established as described here: https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods#enhanced-gotek

Reliably latching motor signals is likely to require external hardware. A 74 flip flop IC perhaps, or small CPLD.

One note on your comments: ULN2803 is unsuitable because Darlington cannot reliably drive TTL low, quite apart from any speed concerns (74HC is already only marginally fast enough hence the diode in the schematic).

TzOk83, 2018-03-04:
I agree that Atari ST floppy connector is rare and expensive, but I really hate mods that require a permanent changes in the computer, like cutting the case. I can accept cutting some bridges on motherboard, but nothing beyond that. You may easily make a floppy swap switch in both STf and STe and mount it without cutting or drilling a case.
The most important mod that I can imagine is populating an existing but not used pads for SPI Flash (U8) with a SPI RAM, and use it to buffer more tracks or even a whole disk image. There are pin compatible SRAM chips which would fit there.

Squirrel61, 2018-03-05:
The hardware mods proposed sound great, but will also require software support, ie. a whole new firmware has to be created. Since the demand will be small, I don't think it's worth making the mods. 

As for the external floppy connectors, I don't know about the Atari but it looks like it will be very hard to get, although it also resembles some industrial connectors that I sometimes find in my job. But for the Amiga, it's pretty easy to convert a DB25 into a DB23. The DB23 female will fit in a DB25 male, although it might be wise to create something that makes it impossible to insert it on the wrong pins, and/or remove the two "spare" pins. And the DB25 female can be converted to fit a DB23 male, by cutting of the last 2 pins & housing. Maybe not the most neat solution, but at least you don't have to destroy your valuable piece of vintage tech.

Sassa0, 2018-03-05:
Thanks keirf, TzOk83 and Squirrel61 for your interest and your valuable comments.

Actually I would not consider my proposal as a new design: at least, it uses all the same pins of the STM32 as the original Gotek. 
Leaving the SD and Amiga things apart, and using a mechanical 2-way switch for drives inversion, it would need no software change : it is almost the same thing as a standard Gotek + all modifications from the hardware mod page. Except that it is thought for internal use + external UI with NO case modification + leaves the mechanical FDD in place.
Because of that, I would rather call it a refitting to the (at least my) needs for retro-gaming.

As you said, of course the demand will stay small compared to the Gotek which is available and easy to go. However enthusiasts who are interested in the boot selection between real and emulated floppy and the almost all internal design may build it. And in addition, if you have the patience to wait for the parts from China, it is more affordable to build 2 yourself (PCB included) than buying 2 Goteks. And you have spare parts ;).

Now about the Amiga.
Since the firmware might someday deal with several emulated drives (at least the 2 that are wired on the Gotek), my idea is to simply be ready for the 4 available on the Amiga ;)
The questionable part on the Amiga is the drive identification. Apart that you need the MTRX signal for that (and maybe also the DRES), I do not believe it would be impossible to do it in software, but it will save external logic and be more versatile to choose the number of emulated drives. And note that it will be needed to emulate the second drive with the standard Gotek... It would then make it possible to just wire it to the external port without any additional logic. Maybe it is the most needed software enhancement in my list ?

About the daisy chain of the mechanical FDD.
As TzOk83 said it can just use a mechanical switch for floppy swap.
On the Amiga, using the mechanical FDD as DF1 would also require both to generate the ID for it, and also to generates its MTR signal. It would be possible to add the HW logic to the board, but it is probably not the most awaited feature anyway? In that case, I then just rather connect it to a free pin of the STM chip, so it might just be possible one day, and to fully control it by software, do the same to its DS signal by the way... (by the way, controlling the DS signal by software would make it electrically safe to swap drives when running, and even toggle a disk change to force redetect).

About the SD card.
I read that the first versions of FlashFloppy used a SD card on a lower end STM chip instead of USB. Maybe this code is still present ? And since most of the pins are available in the Gotek on the flash footprint, maybe it can become an official hardware mod someday ? (most because the only missing pins are the Card detect and Write protect).

About the darlington thing, keirf you are right about the logic level, thank you. I will stay on the 74hc04 or maybe the hc240 for its 8 gates...

TzOk83, 2018-03-05:
The Gotek PCB removed from its case fits the Atari STf (but not the STe, there you have to remove some shielding) case between PSU and FDD.
I've bought a Gotek unit in China for 12€... not sure you can make own board for less.

keirf, 2018-03-14:
This ticket is too open ended, I'm not sure it's the right avenue for this discussion. Better on the FB group perhaps.

closed 2018-03-14 by keirf