keirf/flashfloppy #66: Up-dir button bug

discoHR, 2018-03-04:
I'm using rotary encoder (gray) with twobutton-action = rotary option. When I press up-dir button, I find myself in parent directory but only a single item (..) is diplayed there and rotary encoder doesn't scroll through the list. I'm forced to select that item as the only option which results in skipping the parent directory and going one level too far to parent directory of the parent directory. Only then it's possible to enter the problematic directory (original parent directory) again to list the folders properly.

Steps to reproduce:

1. Create the following directory structure:
   a/b/
2. Place some disk images to b/
3. Enter b/ using rotary encoder and mount an image
4. Eject the image
5. Press up-dir button to go to a/
6. Notice .. is displayed and you can't change it, there are no other items
7. Select the only available item (..)
8. You find yourself in root
9. Select a/ manually (you should be there automatically after step 5)

This bug is present in all builds since rotary = gray was added, could be present earlier too as I haven't tested it before.

discoHR, 2018-03-04:
Here is how OLED looks like when the problem happens. Notice 000/000.

[image]

keirf, 2018-03-04:
Thanks I will repro this and fix.

keirf, 2018-03-05:
Had a play with this following your a/b/<images> structure and could not repro this following your exact recipe.

Please confirm it occurs on v0.9.14a, and if so list your complete USB dir structure and your FF.CFG contents.

discoHR, 2018-03-05:
It occurs on all v0.9.x versions, including v0.9.14a.
My structure is actually a/b/c, please see attached files. Image files are only in deepest directory (c).

[directory-structure.txt](https://github.com/keirf/FlashFloppy/files/1780129/directory-structure.txt)
[FF.CFG.txt](https://github.com/keirf/FlashFloppy/files/1780130/FF.CFG.txt)

keirf, 2018-03-05:
I'll give it another go...

commit 93a0fe83 (2018-03-05): main: Directory cluster in folder stack is 32 bits not 16.
Fixes #66
closed 2018-03-05 by keirf