keirf/flashfloppy #68: DOS ".img" universal recognize (enhancement)

Simone2018, 2018-03-05:
in "img.c"

> bool_t _img_open(struct image *im, bool_t has_iam, const struct img_type *type) 

if correctly formatted an IMG can be recognized by the BPB (first sector): you already have definition in FF.C:

> 79 #define BS_JmpBoot			0		/* x86 jump instruction (3-byte) */ 
> 80 #define BS_OEMName			3		/* OEM name (8-byte) */ 
> 81 #define BPB_BytsPerSec		11		/* Sector size [byte] (WORD) */ 
> 82 #define BPB_SecPerClus		13		/* Cluster size [sector] (BYTE) */ 
> 86 #define BPB_TotSec16			19		/* Volume size (16-bit) [sector] (WORD) */ 
> 87 #define BPB_Media			21		/* Media descriptor byte (BYTE) */ 
> 89 #define BPB_SecPerTrk		24		/* Number of sectors per track for int13h [sector] (WORD)*/ 
> 90 #define BPB_NumHeads			26		/* Number of heads for int13h (WORD) */ 
> 100 #define BS_55AA			510		/* Signature word (WORD) */ "

55AA flag can be checked and if positive all useful info can be get here...

I'm experimenting in very big floppy disk (I have made an .HFE **6.55**Mb big with success in a 720kb controller @100RPM ! but HFE is limited by 65535 cells per track by design, so no more than ~23 sectors/track). Speed can be guessed from the Media descriptor (like F9 = 500.000 bit/s others 250.000 bit/s or better an option in FF.CFG)

Thank you for all your great work!

keirf, 2018-03-14:
Easy to do, though wouldn't make this autodetect a default action. Would need a different file extension than IMG, or enabling via FF.CFG

Simone2018, 2018-03-15:
Thank Keirf, 
yes you are right,
an option like "img_geometry_from_VBR=on" would be fantastic... 
also "Speed=250K" or "RPM=100" would be a great addition. 
Think of a 32 Mb floppy in a 720k drive... you can (slowly) copy entire HD on a single floppy image!

keirf, 2018-06-05:
What would be a good thing to configure this on? Perhaps `host = pc-dos`? The BPB is MS-DOS (PC-DOS) defined right?

Simone2018, 2018-06-05:
Yes, keirf, is a good idea to use **host = pc-dos**. BPB is dos only. 
.DOS file naming is worst IMHO because isn't a standard suffix and other program can't recognize it just by double click.

keirf, 2018-06-06:
Just finalising this. Request for speed/rpm config option would belong in a separate ticket.

commit 41136d4b (2018-06-06): img: New host type "host = pc-dos" scans BIOS Parameter Block in boot sector for image geometry.

Fixes #68
closed 2018-06-06 by keirf