keirf/flashfloppy #69: Unable to format high density on Acorn A30x0/A5000 (possibly riscPC)

drdpj, 2018-03-06:
Low density is working OK. High density will at least run through the format if indexsuppression=no, without this it complains about the drive being empty, but it does completely knacker the format. See attached HFE.
[hdfloppy.zip](https://github.com/keirf/FlashFloppy/files/1786666/hdfloppy.zip)

keirf, 2018-03-06:
High density means 2MB DSHD? Using latest release v0.9.14a?

drdpj, 2018-03-06:
Yes, basically 1.6mb dshd, using latest version.

or mute
the thread

keirf, 2018-03-07:
Okay, the HFE write issue is understood. If the whole track cannot be preloaded then the write path falls back to a super rubbish block-at-a-time mode, which you can expect to take 500ms+ per track. This will not keep up with the host and the write buffer overflows.

So, I can fix that.

But also I see that this is a simple 1024-byte 10 sector-per-track format. It's the same used as on Akai synths for example. So try putting `host = akai` in FF.CFG and use an IMG file instead, and report back how well that works.

drdpj, 2018-03-07:
I'll try that this evening. Will that work with 1024x5 sector tracks too
(at low density)?

d.

keirf, 2018-03-07:
1024x5 will also work, if you use this modified firmware:

[ff_69_1.zip](https://github.com/keirf/FlashFloppy/files/1789389/ff_69_1.zip)

By the way can you link to or explain the naming scheme for Acorn sector images, ADM, ADL, ADF, ... Is IMG ever actually normally used on Acorns?

drdpj, 2018-03-09:
No, sorry, :( I'm going to hopefully try this evening. Juggling too many things at the moment!

drdpj, 2018-03-09:
Just whilst I test, floppy formats for ADFS: 

Suffix|Density|Sectors|SectorSize|Size|Heads
M| Double |16|256|320k|1 *
L | Double | 16 | 256 | 640K | 1 *
-- | -- | -- | -- | -- | --
D/E | Double | 5 | 1024 | 800K | 2

F | Quad | 10 | 1024 | 1.6M | 2

ADF is usually used for the D/E/F formats, but that clashes with the amiga suffix...

*M is single sided, L is double sided, but L fills side 0 first, then side 1 (non-interleaved)

drdpj, 2018-03-09:
With index suppression on, I'm getting intermittant drive empty formatting 800kb sector file - it will let me restart it and it did verify OK.  I'm getting the same with 1.6MB sector, but the format failed saying it was an unknown file object when it finished. I'm not going to get a chance to log it this evening :(

drdpj, 2018-03-09:
OK, formated 800kb HFE ok, but 1.6MB gave drive empty on every track - gave up after about 7 or 8... (I'm thinking the drive empty is to do with the index suppression - I don't get this if I turn it off, strangely)...
[DiskImages.zip](https://github.com/keirf/FlashFloppy/files/1798300/DiskImages.zip)

keirf, 2018-03-09:
Can you try formatting IMGs? Have all these tests been with HFEs? I can tell you now that formatting an HD HFE is doomed, FlashFloppy will be far too slow.

drdpj, 2018-03-09:
Yes, single density IMG (that's in the file I uploaded) formatted OK, but did give me a "drive empty" halfway through, but it was able to continue, finish, and verify OK. The HD IMG (also in the file) did a single "drive empty" but then totally failed at the end of the format, saying it was an unknown file object...

keirf, 2018-03-09:
Ok will take a look :)

keirf, 2018-03-10:
Ok, I messed up in a couple of ways:
1. ADFS sectors are numbered from 0, not 1
2. ADFS D,E,F sectors are skewed across cylinders

I have made you a test branch 'acorn' in the Github repository. Can you please build and test that with IMG files? You need `host=acorn` in your FF.CFG file.

Also, which ADFS double-sided image files are not interleaved? Is it only ADL (ADFS L) files? Did you know that your existing ADL support uses interleaved image files?

drdpj, 2018-03-10:
I'll give this a go later. Thanks :) actually, thinking back, yes I did
make a conscious decision to interleave the sector image as although the
hardware deals with the sides sequentially, all existing images were
distributed as interleaved!

keirf, 2018-03-10:
By the way, if you have a preferred host type name let me know. I wondered if host=adfs better... I get the feeling Acorn is considered to refer to the old 8-bit systems. Either way, this will also allow us to interpret ADF suffix correctly for ADFS D/E/F image files when the host= is specified correctly :)

drdpj, 2018-03-10:
Perhaps acorn-32bit or some convolution of that? Just thinking about
heading off the pedantry that might get wheeled out...

drdpj, 2018-03-10:
OK, img on both gave disk error 23: 
&23 | Sector or ID not found?  The format went ahead though.

keirf, 2018-03-10:
That's mysterious and sounds no better. Gathering serial logs is the best next step I think. Let's concentrate on ADFS D/E (double density) IMG for now, and deal with high density after.

drdpj, 2018-03-10:
OK - remind me how to do a debug build?

keirf, 2018-03-10:
debug=y make -j8 dist

drdpj, 2018-03-10:
OK - I recompiled with the debug build, but now the computer won't boot - I think because the gotek isn't responding quick enough to something? Is that feasible? Works fine with a standard build.

** FlashFloppy v0.9.15a for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Board: Standard
Config: Flash Slot 11 (ver 2, size 48)
Interface: IBM PC
I2C: OLED found at 0x3c
Display: 1602 LCD

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001180927123570
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
0:F: 'emptySD.img'
Mode: Native
Current slot: 8/8
Name: 'emptySD' Type: img
Attr: 20 Clus: 0000480c Size: 819200
New max: read_us=1143
New max: read_us=1256
New max: read_us=1267
New max: read_us=1295
New max: read_us=1565
New max: read_us=1575
New max: read_us=1586
New max: read_us=1594
New max: read_us=1610
New max: read_us=1628
New max: read_us=1798
New max: read_us=1806
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit

drdpj, 2018-03-10:
OK - long log now - it's doing something stupid with the sector IDs (I had to reduce the speed back to 115200): ** FlashFloppy v0.9.15a for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Board: Standard
Config: Flash Slot 11 (ver 2, size 48)
Interface: IBM PC
I2C: OLED found at 0x3c
Display: 1602 LCD

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 4C530001180927123570
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
0:F: 'emptySD.img'
Mode: Native
Current slot: 8/8
Name: 'emptySD' Type: img
Attr: 20 Clus: 0000480c Size: 819200
New max: read_us=1110
New max: read_us=1111
New max: read_us=1246
New max: read_us=1324
New max: read_us=1437
New max: read_us=1553
New max: read_us=1606
New max: read_us=1665
New max: read_us=1784
New max: read_us=1797
I2C: Error (0000)
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
[... 910 lines cut ...]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]
IMG Bad CRC 5dfb, sector 0[1]
IMG Bad CRC 5dfb, sector 1[2]
IMG Bad CRC 5dfb, sector 2[3]
IMG Bad CRC 5dfb, sector 3[4]
IMG IDAM Bad Sector: 0
IMG Bad CRC 5dfb, sector 4294967294[4294967295]

keirf, 2018-03-10:
You are missing config=acorn from your FF.CFG file, or it is not being interpreted correctly.
1. You are building and testing the `acorn` branch, right?
2. You should see ff_cfg.host == HOST_acorn. You could check via printk()
3. This should be picked up in img_open() and cause the adfs_type[] array to be used.
4. Therefore you should find in _img_open that you end up with a 5*1024-byte geometry with sector id base 0. You could check that via printk() too.

drdpj, 2018-03-10:
Knickers, that was the wrong branch (git ignorance on my part) - OK, 800KB formats and verifies fine,  1600kb formats and verifies fine. Excellent!

d.

keirf, 2018-03-10:
Okay, this is merged into master, could you do a final kick of the tyres?

This time your images should have the proper ADF suffix. The IMG suffix will not work!

Apart from that, as before, just make sure you build/test latest `master` branch ;)
The FF.CFG option remains `host = acorn`

drdpj, 2018-03-10:
Can I just check what suffix you're expecting for High/Double density for the Arc formats? ADF for both?

d.

drdpj, 2018-03-10:
(it's not recognising ADF files now :D)

keirf, 2018-03-10:
My mistake, forgot to fix image_valid()

keirf, 2018-03-10:
Pull latest master and try again

drdpj, 2018-03-10:
Getting there - 800kb formats and I was able to write a 320k file no problem.  1600kb formats, verifies, but gave me a drive empty partway through saving a 640kb file... :(

keirf, 2018-03-10:
See how you get on with `index-suppression = no`

drdpj, 2018-03-10:
that just made it stutter about the tracks (it worked) but it was very slow. I've just tried it with real-time on track-change, that seems better, but I have had the odd spurious drive empty - although it's not consistently repeatable. Something must be marginal...

drdpj, 2018-03-10:
Double density is fine - I can't make that barf. I've tried a few times with big saves. I'm assuming it's not quite keeping up in high density mode? 

 Also worth noting that the arcs will read IBM format discs (360-1.44) and ST format (720/360).

keirf, 2018-03-10:
DD fine with `index-suppression = yes`?

Does the Archimedes expect a Shugart or IBMPC floppy interface by the way (is pin 34 /RDY or /DSKCHG)? And is that configured right?

drdpj, 2018-03-10:
IBM - Pin 34 is /DSKCHG - and yes, DD is fine with index-suppression = yes.

keirf, 2018-03-10:
And you configured that by jumper on JC or by `interface = ibmpc`?

drdpj, 2018-03-10:
Jumper on JC.

keirf, 2018-03-10:
Ok and have you tried a non-debug build? 

A bit of googling suggests A5000 does require regular index pulses (http://www.riscos.com/support/developers/prm/adfs.html ; Index - Input), which is a bit of a pain.

drdpj, 2018-03-10:
Yes, this is with the non debug build... :-/

keirf, 2018-03-10:
Then debug build and grab a log of a write ending in "drive empty" and let's see if that shows anything.

drdpj, 2018-03-10:
Okay, here you go:
[HDcopies.log](https://github.com/keirf/FlashFloppy/files/1799723/HDcopies.log)

keirf, 2018-03-10:
Okay clearly latency suddenly increases at the end of the log, and that's probably the cause. It could be caused by the USB stick starting garbage collection in the FTL. Perhaps try another stick and see how that goes?

Even more concerning are the "IMG Bad CRC" messages, as these are failed writes (fail data CRC check).  Probably worth checking if a large write to a DD image has these, and/or validate the file after writing it to make sure the data is all there!

CRC errors need to be solved before anything else.

drdpj, 2018-03-10:
Double density log attached - no problems there! :) (ignore filename :D)

d.
[singledensity.log](https://github.com/keirf/FlashFloppy/files/1799886/singledensity.log)

keirf, 2018-03-11:
Thanks, I realise now the issue is that you are logging at 115200 baud and this is causing missed deadlines because logging occurs with interrupts disabled. Average log line approx 25 chars @ 10 bits per char @ 1/115200 sec per char ~= 2.5ms.

WDATA capture buffer is 1024 entries, each flux change is 2-4us for HD data, hence buffer overflows in 2-4ms.

A shame you can't log at higher rates, but maybe that is unsupported by your logging software? The PL2303 clone USB-TTL adapter hardware certainly supports it.

keirf, 2018-03-11:
Got a draft plan for hopefully better implementation of `index-suppression = no` here: https://github.com/keirf/FlashFloppy/issues/67#issuecomment-372115791

keirf, 2018-03-12:
Could you try building and testing branch `fake-index` with index-suppression=no, please? Alternatively you can grab a non-debug version already built from this link: https://github.com/keirf/FlashFloppy/files/1802577/ff_67_3.zip

keirf, 2018-03-12:
Scrub that, didn't work on #67. Will try a different approach tomorrow.

keirf, 2018-03-13:
Okay I have a couple more firmwares to try with `index-suppression=no`, as follows:

ff_67_7: Injects extra spurious INDEX pulses at end of host writes and on every head step. Also repositions read head at end of track (to generate another INDEX pulse) after every host write. Hence you may find performance of this one is bad for multi-sector writes. Hopefully no "drive empty" errors though:
https://github.com/keirf/FlashFloppy/files/1806493/ff_67_7.zip

ff_67_6: Injects the extra INDEX pulses only at the end of host writes. Does not reposition the read head (so multi-sector writes will go at full speed). More risk of "drive empty" errors, should go fast though:
https://github.com/keirf/FlashFloppy/files/1806007/ff_67_6.zip

drdpj, 2018-03-13:
OK. I shall try this evening!

drdpj, 2018-03-13:
HFE double density, Sector Double Density (800kb) and Sector High Density (1600kb) seem to be fine! Excellent work! :) :)

drdpj, 2018-03-13:
(This was writing a massive 1.5mb file on the high density disk, seemed as fast as a normal floppy)

keirf, 2018-03-13:
That's somewhat surprising but very good news :) I will check up on #67.

keirf, 2018-03-14:
Fix will be in v0.9.16a

closed 2018-03-14 by keirf

keirf, 2018-03-16:
FYI, v0.9.16a is now released. Please do test and confirm it still works happily on the Archimedes. If there are still issues (as there do seem to still be on BBC 8271/DFS systems, as you're aware) then please do open a fresh ticket to address that.