keirf/flashfloppy #70: 2.88 disk images, LS120/240, ZIP100, ZIP250

linuxmaster, 2018-03-10:
Is it possible to implement support of 2.88 disk images, LS120/240, ZIP100, ZIP250? How we can help you with that?

keirf, 2018-03-10:
2.88 ED is already "supported" in that an IMG file of the appropriate size will have its geometry auto-detected. So first step is to try an existing ED IMG and see if you can read that okay via FlashFloppy. If not then serial logs from a debug build will be useful.

The data rate of the various super floppy formats is likely too high to be supported by full-speed USB. I think a board based around STM32F4 with HS USB and more RAM for buffering would be needed.

TzOk83, 2018-03-10:
Neither SuperDisk nor ZIP drives were using Shugart-like interface. These were IDE drives (in external variant - ECP/LPT or USB).

keirf, 2018-03-10:
Ticket restricted to 2.88 ED disk images (IMG files). Left open for now in case you would like to test this support.

keirf, 2018-03-14:
Pleease reopen if you wish to test 2.88MB ED

closed 2018-03-14 by keirf