keirf/flashfloppy #71: Create blank floppy

smartroad, 2018-03-10:
Just a thought, would it be possible to get flashfloppy to be able to make a blank disk?

My idea would be when in directory browsing mode, you could maybe hold the eject button for a few seconds and it creates a blank floppy based on a setting in FF.cfg - the setting would hold both the naming structure FLOPPYxx for example and the type of file to creat, add, hte etc.

May not be possible and may be viewed as pointless, but would be a neat way to be able to copy ADF's on my Amiga to a drive to be used without having to turn on my main PC to copy over to a single image.

Just a thought :)

keirf, 2018-03-10:
I will give it a thought. I'll keep the ticket open for now.

Simone2018, 2018-03-10:
Maybe a function to duplicate active disk image would be easier... so I select blank image from my own directory and a new copy of the same floppy with different name is saved. Useful also if you want to do some experimental write and keep original image.
I agree for simple command like "keep pushing rotary button" like eject for more than xx seconds...

tkurbad, 2018-05-23:
I would also greatly appreciate a floppy create feature.

tenuke, 2018-05-23:
But it takes just few seconds to multfiply sny image on pc blank or full
and full you can format then on the machine you use ff with

smartroad, 2018-05-23:
Tenuke, that assumes you have a PC switched on, otherwise I'd have to switch it on, wait for it to boot, copy/transfer, then switch off again. Kinda a waste of power just to make one blank disk :D Or what if your pc is in another room or broken ;)

drdpj, 2018-05-23:
Possibly a neat solution would be to select a default blank image size in bytes in ff.cfg and then have a particular button press that creates blank-n.whatever in the current directory where n is a number.  You can always rename them at a later date when you do have a bigger machine fired up.

RetroClinic, 2018-05-23:
Thing with that is the BBC Micro can have multiple different file sizes, 200K for SSD, 400K for DSD, 320K for ADFS M, 640K for ADFS L (as I’m sure other systems can). It would need an additional step to select the file size you want from a set of available image sizes in the CFG file, that would be useful.

From: DanielJ 
Sent: Wednesday, May 23, 2018 2:44 PM
To: keirf/FlashFloppy 
Cc: Subscribed 
Subject: Re: [keirf/FlashFloppy] Create blank floppy (#71)

Possibly a neat solution would be to select a default blank image size in bytes in ff.cfg and then have a particular button press that creates blank-n.whatever in the current directory where n is a number. You can always rename them at a later date when you do have a bigger machine fired up.

bjernler, 2018-10-28:
I would love that function. I am currently backning up floppys to the gotek, and it is a pain to duplicate, rename, copy... Then start over

pipzuk, 2019-01-27:
I'd also go with the "Copy" feature as it ticks all the boxes, it gives a true copy function to allow messing with images without corrupting the original, a blank image could be stored on the USB and copied to give a new blank floppy - and if the BBC uses a variety of formats just pre-creating one of each on the USB would make all available later, then it's just the naming convention that's the headache (especially for 8.3 file systems). Currently I have a bank of "blank" floppies on my USB and if I use one I rename it later as suggested above but it does clog the directory up somewhat!

keirf, 2019-06-13:
v3.1a includes a Clone Image operation in the new "Eject Menu" for LCD/OLED displays when an image is ejected.

The Clone operation clones an image.suffix to image_000.suffix, image_001.suffix, and so on. The selected clone number is always the highest existing clone number, plus 1.

I believe that satisfies this ticket. Please try it out and raise issues as necessary.

closed 2019-06-13 by keirf

DOSputin, 2019-12-02:
Been trying to clone blank `.HFE` images on an E-mu Emulator II.  
I have my usb disk prepared, and a blank image named `blank.hfe`    
From the eject menu, I select the `blank.hfe image, navigate to `Clone Image` and select `Confirm Clone`.  
At this point nothing appears to happen, and I am returned to the `Clone Image` menu entry.  

Is there anything that I am missing?

juj, 2021-07-20:
Could this ticket be reopened?

There may exist a Clone Image option - but I don't have any image files on my USB, it is empty. It would be very flexible to be able to just choose "Create New Disk -> 360KB/720KB/..." from the menu to populate an empty USB. In my use case I would have liked to create a new bootable DOS disk via "format A: /S" or "sys A:" on an actual DOS machine to a blank USB drive, but looks like that is not possible at the moment - I am just stuck with 000/000 screen.

gingerbeardman, 2022-02-20:
Also here for create blank.

keirf, 2022-02-20:
I don't plan to implement create since there is already a clone (copy/paste) function. Any blanks you would like supplied (or to supply) belong in the flashfloppy_image_library repository.

gingerbeardman, 2022-02-20:
https://github.com/keirf/flashfloppy-images

keirf, 2022-02-20:
That's the one. It's a bit empty. But I'm happy to host blanks of all kinds there!