keirf/flashfloppy #72: Amstrad CPC6128 program loading issue

Krisu13, 2018-03-13:
I tried to use GOTEK with FlashFloppy (9.15A) software to load some games to my Amstrad CPC6128.
I got this error message [https://drive.google.com/file/d/1ZYCRSUUh4Z0uCa0cWmoMbhm5ZOkOKzoY/view?usp=sharing](url)
I used the same GOTEK drive with Amiga and Atari worked fine. 
Just to make sure that it is not a connection (cable) issue I swapped Gotek drives for one with HxC software and it worked fine. I was able to load games.  
Any idea what I’m doing wrong or is it a software bug? 
Thank you 
Kris

keirf, 2018-03-13:
Not tried the CPC file selector in a long while, most CPCers are using Zaxon's designs with LCD/OLED and direct navigation on the device. I will test this myself before doing the next release.

keirf, 2018-03-13:
I assume you used the same USB stick with same files on it on both Goteks?

Krisu13, 2018-03-13:
Thank you for quick reply. 
Yes I used the same mem stick on both devices.
I will try your suggestions and will provide link to AUTOBOOT.HFE tonight.

Krisu13, 2018-03-14:
Keirf

when I downgraded to 9.6a everything worked fine. 
here is the link to AUTOOBOOT.HFE https://drive.google.com/file/d/1RC4B0TzAkkrGOdsZYFjWMT335cNDS4Cg/view?usp=sharing

Great project - thank you 

Kris

keirf, 2018-03-14:
Thanks, reproduced this failure myself. The bug was introduced in v0.9.14a by commit 4e9a91126ff4542823926b2f2b64e7a8b28f3d39

keirf, 2018-03-14:
Great bug, thanks for the report. This is fixed in master and will be rolled into the next release (v0.9.16a).

closed 2018-03-14 by keirf