keirf/flashfloppy #73: Feature Request: usb external floppy support

soydemadrid, 2018-03-14:
Hi I don't know if I've posted in the right place for a feature request pr if this is somehow already possible or planned?

BUT, a great idea would be to allow an external USB floppy disk drive to be plugged into the gotek usb port.

That way if you swap a floppy drive with a gotek in your old hardware due to a broken drive you can also buy a cheap usb external floppy to still access original disks from computers that wouldn't normally support usb or a usb pc floppy external (like amigas or amstrad cpc etc).

For example I have an Amstrad PPC which has two dsdd internal floppies. It would be awesome to swap for two internal goteks and then just plug my cheap external usb PC floppy into one of the gotek usb ports to copy original disks to the other one. Then just unplug it and go back to gotek only again when stuff is moved over.

Also on my amiga 1200 it would mean the gotek can be the internal drive and original floppies can still be used easy and quick via usb when needed.

How cool would that feature be if possible?

The external PC usb when plugged in could just automap to the first gotek track 001 or something like that...

keirf, 2018-03-14:
USB floppy drives aren't as flexible as bare drives. They can certainly only read IBM-sectored disks (so AmigaDOS disks are out of the question, for example). They may even be restricted to the few formats needed for IBM-PC (eg. 18spt 1.44MB HD).

Anyway, regardless, it is outside the remit of FlashFloppy I'm afraid.

closed 2018-03-14 by keirf

soydemadrid, 2018-03-14:
Ok thanks. That is a shame, even if it would work for dos/pc formats only it would of been awesome. But thanks for replying and looking at they idea anyway.   :)