keirf/flashfloppy #74: SJ Research MDFS File Server

Ottly, 2018-03-17:
Hi Keirf,

My SJ MDFS File Server was kinda working with Flash Floppy up until firmware 9.11a after that it is not working at all. When I say kinda working it was booting from the Gotek and I could read the disk image. Writing was an issue though. Disk formats would fail and writing to a working HFE image would eventually fail (for example when deleting files that exist on the disk image).

Attached is a MDFS Floppy disk image I created using HxCFloppy Emulator v2.1.8.0 (Floppy Disk Dump).

Let me know what else you require from me to help track this bug down.

Thanks, Scott.

[MDFS Master FS Ver 2.zip](https://github.com/keirf/FlashFloppy/files/1821276/MDFS.Master.FS.Ver.2.zip)

keirf, 2018-03-17:
So "kinda" worked up to an including 0.9.11a? And then stopped working at all at 0.9.12a?

keirf, 2018-03-17:
Okay I see SJ MDFS is an Acorn Econet add-on. Reasonable to guess that it may have the same requirements on index timings as some other Acorn drives. So, here are two things to try, both requiring an [FF.CFG](https://github.com/keirf/FlashFloppy/wiki/FF.CFG-Configuration-File) text file in the root of your USB stick containing:
```
index-suppression = no
host = acorn
```

The first line above inserts INDEX pulses during seek operations (PC HD drives suppress them during seek). Some Acorn systems expect regular index pulses or they consider drive empty.

The second line above will let you use sector image files instead of HFE files. That is an 800kB file containing sectors of track 0 side 0, track 0 side 1, track 1 side 0, ...
The file should have the suffix/extension *.ADF (std extension for ADFS D/E/F images).

Ottly, 2018-03-18:
Yes works with 0.9.11a and earlier. I started with 0.9.6a. From 0.9.12a onwards, not working.

ADFS disks wont read in the MDFS as its a slightly different format and vice versa, MDFS Floppy Disks can't be read on Acorn BBC computers.

However I tried to format a ADFS 640kb disk image in the MDFS and it formatted successfully (Flash Floppy 0.9.16a with index-suppression = no and host =acorn).

MDFS floppy disks are 800k so I experienced corruption once I went over 640k writing to the floppy image. Will try with a 800k .ADL disk image and report back.

keirf, 2018-03-18:
The only significant relevant change in 0.9.12a is fixing the head-settle time emulation, so it takes ~10ms longer to serve up track data after a seek. I guess long enough for INDEX-pulse timeout in the MDFS hardware. Sounds like you're nearly there -- the 800k sector image ought to work (geometry 80 track, 2 sides, 5 * 1024-byte sector per track).

PeterBBCUser, 2018-03-18:
does switch 3 and 4 (both on) on the keyboard also speed up access time on MDFS as I found on the BBC images it helped with having a faster seek time otherwise it would time out if doing long seeks.
Peter.

keirf, 2018-03-18:
@PeterBBCUser incidentally it may not make any difference any longer as I insert an index pulse on every step pulse now, so plenty enough to keep the time-out circuitry from triggering.

Ottly, 2018-03-20:
ok ,i've been doing some testing. Firstly .ADF extension images will only display on the GoTek OLED if host = acorn. If host=unspecified the disk images don't show at all. I'm just mentioning this as I thought the Amiga also used that filename extension and hence should still get displayed?

The MDFS seems to work fine with index-suppression = yes. Also works with  index-suppression = no.

With host = unspecified or host = acorn a 800kb .ADL disk image appears to work fine. Can format, copy files until disk is full, delete files etc.. 

Is there any problem with using .ADL extension that I should be aware of?

With host = acorn a 800kb .ADF disk image appears to work fine. Can format, copy files until disk is full, delete files etc.. If host = unspecified then the disk image won't appear at all.

Does not like .HFE disk images at all.

I'm happy for this ticket to be closed as I have a working solution.

Thanks again Keirf. Happy to do more testing if required.

Scott.

keirf, 2018-03-20:
Okay I looked at your HFE a little closer and the header states that this is a 360RPM image. This is probably why it doesn't work, as it's much more likely that MDFS uses 300RPM drives. As of 0.9.13a HFE handling more accurately deals with uncommon track lengths and rotation rates -- and this presumably broke your HFE images with incorrect header infos.

So for the record, HFE will probably work fine for you if it states 300 RPM (data rate 250000 bit/s) in the header. And you definitely don't need `index-suppression = no`.

Good to hear the sector images work fine. The ADFs not appearing if `host = unspecified` makes sense as Amiga ADFs are a multiple of 11*512 bytes and the image-validity check filters on that.

closed 2018-03-20 by keirf

Ottly, 2018-03-21:
You were spot on with the .HFE image. It also works fine now if at 300 RPM and data rate 250000 bit/s.

Just an FYI. The MDFS has an MB8877A Floppy disk controller.

Thank you so much again for all your work on flash floppy Keirf. The GoTek really is a game changer for retro hardware.