keirf/flashfloppy #75: Spurious index

claunia, 2018-03-20:
Sometimes on read or writes there's a spurious index pulse before the full 200ms, and the 200ms timer resets after this spurious pulse. (checked with a logic analyzer)

keirf, 2018-03-21:
Please attach a (Saleae?) trace file. Or at least attach a screen capture.

Index pulses aren't guaranteed to be 200ms apart on FlashFloppy, however they would rarely be *closer* than 200ms, except when `index-suppression = no`. So worth a quick look.

claunia, 2018-03-21:
Oh I set that config.

I can only generate "sigrok" files :/

Or a capture but well, in the /INDEX line, in between the 200ms pulses, there is a pulse every 6ms

keirf, 2018-03-21:
Don't set `index-suppression = no`, it's only needed for some Acorn systems which use regular index pulses to detect disk-inserted. However, with that option you will see extra 'fake' pulses on every head step and immediately after every write.

I think sigrok format is okay, there's free software to view them I believe?

claunia, 2018-03-21:
indeed whole libsigrok, dunno if for windows but it's what I used for creating the dump on Linux

keirf, 2018-03-22:
I use Linux anyway so not a problem.

keirf, 2018-03-25:
This will be a symptom of `index-suppression = no`. Addressed on other tickets. Closing.

closed 2018-03-25 by keirf