keirf/flashfloppy #76: Formatting issues

claunia, 2018-03-21:
Ok as we talked about, I finally got the logic analyzer connected to the gotek and found the following:

Socket 775 Super I/O cannot format at all, not even on real floppies, they never enable WGATE or WDATA signals.

So took a Socket A and got mixed results:

- With .IMG, FreeDOS FORMAT.COM and Linux fdformat work correclty.
- With .HFE both of them fail on verifying the image.
- With .HFE + FDFORMAT (DOS) it goes directly to track 81 and returns drive not ready error.
- With .HFE + 2M, it formats, and starting at track 5 or 6 (not always the same) it recalibrates for verifying, but verifies correctly,then at track 80 it recalibrates but returns to track 79 not track 80 failing format there. Checking with ANADISK, the tracks previous to that appear correctly formatted.
- With .HFE + 2MGUI, it gives drive not ready after 3 tracks.

So, it can work, but seems to have some timing issues right now.

keirf, 2018-03-21:
In all cases trying to format a high-density HFE? This is no surprise at all since writeback strategy for a track that does not fit entirely in RAM is to do it block by block, which is horribly slow.

If you are interested I can make a patch for you to try which does batch-at-a-time.

claunia, 2018-03-21:
Absolutely interested @keirf remember that my usage scenario is format .hfe in all formats I can :stuck_out_tongue:  please

claunia, 2018-03-21:
ok seems that double density is working more or less correctly, now if HD and ED could be working it would be just perfect.

keirf, 2018-03-22:
I doubt that ED HFE is possible. Full-speed USB can't maintain that data rate. ED IMG though, yes, maybe possible. You will have to try it.

claunia, 2018-03-22:
We can try, anyways I can live without, but HD is a must. Also I've found some formatters have problems with DD mode also (incorrect density in 2M, address mark not found in FDFORMAT). I'll prepare a complete table today or tomorrow (with partially written HFE available at your request if needed).

claunia, 2018-03-25:
the problem with 2M and possibly the same problem with FDFORMAT is that according to the logic analyzer, pin2 gets set to high density no matter the disk image. I tried with a real floppy (without the logic analyzer there) to discard a fdc problem because everywhere I read it says that pin should be set by the controller. But with a real floppy they work.

claunia, 2018-03-25:
With an SSD connected seems that HD formatting works more or less ok (2M still gives an error on last track and FDFORMAT gives drive not ready on last track). Indeed, it works better than DD.

http://www.claunia.com/logicff.tar

`logic/fdformat800.sr` is a capture of FDFORMAT A: /F800 with MF2DD.HFE (seeks to t79 and returns drive not ready error)
`logic/2mguihd.sr` is a capture of 2MGUI with MF2HD.HFE (partial)
`logic/2mguidd.sr` is a capture of 2MGUI with MF2DD.HFE (tries to format and returns incorrect density error)
`logic/2mfhd.sr` is a capture of 2M with MF2HD (partial)
`logic/2mfdd.sr` is a capture of 2M with MF2DD (tries to format and returns incorrect density)

The following only capture /REDWC line
`logic/2mfddonhd.sr` is a capture of 2MF A: /DD with a real floppy drive and an HD disk inserted (returns incorrect density)
`logic/2mfddondd.sr` is a capture of 2MF A: /DD with a real floppy drive and an DD disk inserted (works)
These two are, well, I'm not sure they are useful, if you need a capture of all the lines tell me.

claunia, 2018-03-25:
Ok @keirf one step at a time, so split bug in #83, #84, #85, #86 and #87, closing this and we attack one bug at a time.

closed 2018-03-25 by claunia